capacitor-community/camera-preview #2: CameraPreview does not have web implementation.

hugovisk, 2019-08-19:
Hi, Tried to run the plugin, but got this errors on console:

> ERROR Error: Uncaught (in promise): CameraPreview does not have web implementation.
>     at resolvePromise (zone-evergreen.js:797)
>     at Function.reject (zone-evergreen.js:889)
>     at CapacitorWeb.push../node_modules/@capacitor/core/dist/esm/web-runtime.js.CapacitorWeb.pluginMethodNoop (web-runtime.js:33)
>     at HomePage.ngOnInit (home.page.ts:16)
>     at checkAndUpdateDirectiveInline (core.js:27788)
>     at checkAndUpdateNodeInline (core.js:38472)
>     at checkAndUpdateNode (core.js:38411)
>     at debugCheckAndUpdateNode (core.js:39433)
>     at debugCheckDirectivesFn (core.js:39376)
>     at Object.eval [as updateDirectives] (HomePage_Host.ngfactory.js? [sm]:1)

-------
**Created a blank app:** 
ionic start cameraPreview blank --capacitor

**instaled the plugin:**
npm i capacitor-camera-preview

**instaled pwa-elements:**
npm i @ionic/pwa-elements

**put the start camera preview instance on home.ts:**
import { Component, OnInit } from '@angular/core';
import { Plugins } from '@capacitor/core';

const { CameraPreview } = Plugins;

@Component({
  selector: 'app-home',
  templateUrl: 'home.page.html',
  styleUrls: ['home.page.scss'],
})
export class HomePage implements OnInit{

  constructor() { }

  ngOnInit() {
    CameraPreview.start();
  }
}

**than:**
ionic serve

----

**ionic Info**
Ionic:

   Ionic CLI                     : 5.2.5 (/home/user/.npm-global/lib/node_modules/ionic)
   Ionic Framework               : @ionic/angular 4.7.4
   @angular-devkit/build-angular : 0.801.3
   @angular-devkit/schematics    : 8.1.3
   @angular/cli                  : 8.1.3
   @ionic/angular-toolkit        : 2.0.0

Capacitor:

   Capacitor CLI   : 1.1.1
   @capacitor/core : 1.1.1

Utility:

   cordova-res : not installed
   native-run  : not installed

System:

   NodeJS : v12.8.1 (/usr/bin/node)
   npm    : 6.10.3
   OS     : Linux 5.0

------
Am I missing something?

arielhernandezmusa, 2019-08-19:
Hi, try importing plugin's module on app.module.ts

import 'capacitor-camera-preview'

arielhernandezmusa, 2019-08-19:
Adding  import 'capacitor-camera-preview',

in html add

<div id="cameraPreview"></div>

and 

ngOnInit() {
    const { CameraPreview } = Plugins;
    CameraPreview.start({ parent: "cameraPreview"});
  }

it will work

arielhernandezmusa, 2019-08-19:
Maybe I have to improve documentation

arielhernandezmusa, 2019-08-19:
Also yo can pass `{.. className: 'yourClassName'}` to customize camera layout

hugovisk, 2019-08-20:
Hi there,

It worked as you described!

My bad ... I misunderstood the setup, I ended up associating with the capacitor camera plugin setup.

I really appreciate your time, help and quick response, thank you very much.

closed 2019-08-20 by hugovisk

arielhernandezmusa, 2019-08-20:
Thanks to you, I will improve documentation to avoid setup problems.

dev-itzarun, 2024-12-25:
> Also you can pass `{..., className: 'yourClassName'}` to customize camera layout

@arielhernandezmusa 

 Is is possible to place camera preview inside a container div in web (svelte / react)
 
 ``` CameraPreview.start({ parent: "cameraPreview"}); ``` this not working with my tailwindcss container