capacitor-community/camera-preview #5: Is Android support ready to use?

jesobreira, 2019-08-30:
I see that commit #29ebeb5 aims to have added Android support.

However on my tests, it only works on iOS. Under Android, the plugin is not available under @capacitor/core/Plugins namespace.

```
import { Plugins } from "@capacitor/core"
const { CameraPreview } = Plugins

console.log(CameraPreview, Plugins)
CameraPreview.start()
```
[image]

Am I missing something? Or Android support is still under development?

Thank you!

arielhernandezmusa, 2019-09-04:
Did you add the plugin to MainActivity?

jesobreira, 2019-09-04:
Thank you for your response!

What should be added to MainActivity?

arielhernandezmusa, 2019-09-05:
You need to add 

add(CameraPreview.class);

To the init method

jesobreira, 2019-09-07:
Thank you!

Is this disclosed on the README or some plugin-specified docs? If not, may I send a PR?

Going a little out of the issue subject, is there support for switching the front/rear camera?

arielhernandezmusa, 2019-09-08:
In the capacitor’s documentation is explained how to used the plugins, but I believe that I have to include it on documentation.

Switching camera is not supported right now, a PR is welcome ;-)

closed 2019-09-27 by arielhernandezmusa