capacitor-community/camera-preview #7: Ios

elepetit, 2019-10-03:
Hello,

On Ios, the plugin starts well but unable to view the camera.
My css include a background transparent.

Thx for your work !

arielhernandezmusa, 2019-10-03:
Hi, can you provide more details, such as iOS version, device ...

elepetit, 2019-10-03:
Hey,

That's ok now. I've to put

` ion-content { --background: transparent !important; }`

in my scss page for Ionic4
but It seems that front doesn't work.
I've got only the rear camera.

arielhernandezmusa, 2019-10-03:
Did you try flip() it?

elepetit, 2019-10-03:
Iphone 7
Ios 12.4.1
Ionic4

arielhernandezmusa, 2019-10-03:
OH, unfortunately, I don't have an iPhone with ios 12.4.1, I just test it on iPhone 7 plus ios 13 and its work

arielhernandezmusa, 2019-10-03:
If you find the problem please, create a pull request with a fix

elepetit, 2019-10-03:
tried to flip, CameraPreview.flip(); but nothing works

elepetit, 2019-10-03:
the swift code looks good.. hum hum ! where's the pb on ios 12....

elepetit, 2019-10-04:
Hi,

I've found the problem.
You were not handling the position option.
I've modified Plugin.swift and CamaraController.swift and that's ok

elepetit, 2019-10-04:
unfortunately ompossible to create a pull

arielhernandezmusa, 2019-10-04:
Ok, can you send ti me the changes you made?

elepetit, 2019-10-04:
in Plugin.swift

`@objc func start(_ call: CAPPluginCall) {
**//add getter for position**
        guard let camera_option = call.options["position"] as? String else{
            call.reject("Must provide position")
            return
        }
        DispatchQueue.main.async {
            if (self.cameraController.captureSession?.isRunning ?? false) {
                call.reject("camera already started")
            } else {
**//send camera_option as camera_position**
                self.cameraController.prepare(camera_position: camera_option){error in
                    if let error = error {
                        print(error)
                        call.reject(error.localizedDescription)
                        return
                    }
                    self.previewView = UIView(frame: CGRect(x: 0, y: 0, width: UIScreen.main.bounds.size.width, height: UIScreen.main.bounds.size.height))
                    self.webView.isOpaque = false
                    self.webView.backgroundColor = UIColor.clear
                    self.webView.superview?.addSubview(self.previewView)
                    self.webView.superview?.bringSubviewToFront(self.webView)
                    try? self.cameraController.displayPreview(on: self.previewView)
                    call.resolve()

                }
            }
        }
    }`

elepetit, 2019-10-04:
in CameraController

`func configureDeviceInputs() throws {
            guard let captureSession = self.captureSession else { throw CameraControllerError.captureSessionIsMissing }
            
            **if camera_position == "rear"**{
                if let rearCamera = self.rearCamera {
                    self.rearCameraInput = try AVCaptureDeviceInput(device: rearCamera)
                    
                    if captureSession.canAddInput(self.rearCameraInput!) { captureSession.addInput(self.rearCameraInput!) }
                    
                    self.currentCameraPosition = .rear
                }
            }
            **else if camera_position == "front"**{
                if let frontCamera = self.frontCamera {
                    self.frontCameraInput = try AVCaptureDeviceInput(device: frontCamera)
                    
                    if captureSession.canAddInput(self.frontCameraInput!) { captureSession.addInput(self.frontCameraInput!) }
                    else { throw CameraControllerError.inputsAreInvalid }
                    
                    self.currentCameraPosition = .front
                }
            }
            else { throw CameraControllerError.noCamerasAvailable }
        }`

arielhernandezmusa, 2019-10-04:
Perfect, thank, I'll apply these changes soon.

elepetit, 2019-10-05:
Ok
I’m working on options

BrunoTCouto, 2019-10-08:
Hi ! I'm having the same issue ! any prediction when this chnges will be merged ?

arielhernandezmusa, 2019-10-08:
Hi, I was on vacations so maybe tomorrow I fix it

BrunoTCouto, 2019-10-08:
Great ! thanks for the update !

arielhernandezmusa, 2019-10-08:
@BrunoTCouto , @elepetit the fix is done :-) thanks @elepetit for your help

closed 2019-10-08 by arielhernandezmusa

BrunoTCouto, 2019-10-08:
@arielhernandezmusa Thanks ! Going to test the fix!

BrunoTCouto, 2019-10-08:
Hm, looks like my issue is not exactly related with this one. Will be opening an new issue with more details

commit 750e910b (2019-10-08): Fixes #7 regarding to camera option