capacitor-community/camera-preview #8: [iOS] - camera already started

BrunoTCouto, 2019-10-08:
Hi,
I'm trying to open the camera in my iPhone but i'm receiving the following error "camera already started", but its the first time the camera is starting...
Not sure what is causing this issue... 

iPhone x Max
iOS  13.1

by the way, the plugin is working fine in the browser :)
thanks for the great plugin !

BrunoTCouto, 2019-10-08:
And the project is runing in Vue with Typescript

BrunoTCouto, 2019-10-08:
[image]

BrunoTCouto, 2019-10-08:
[image]

elepetit, 2019-10-08:
Hi Bruno

Can you provide full code of opening the camera?

BrunoTCouto, 2019-10-08:
```ts
import { Component, Prop, Vue } from 'vue-property-decorator';
import { MediaCameraValidation } from './media-camera-validation.interface';
import 'capacitor-camera-preview'
import { Plugins } from '@capacitor/core'

const { CameraPreview } = Plugins
 
@Component({})
export class MediaCameraComponent extends Vue {

    @Prop({ default: 'enviroment', type: String })
    public facingMode: string;

    @Prop({ default: 20, type: Number })
    public preferredMaskPaddingTop: number;

    @Prop({ default: 300, type: Number })
    public preferredMaskWidth: number;

    @Prop({ default: 345, type: Number })
    public preferredMaskHeight: number;

    @Prop({ default: [], type: Array })
    public validations: MediaCameraValidation[];

    @Prop({ default: false, type: Boolean })
    public faceValidation: boolean;
    
    public containerHeight: number = window.screen.height;
    public containerWidth: number = window.screen.width;
    public containerZoom: number = 100;
    public maskWidth: number = 0;
    public maskHeight: number = 0;
    public maskPaddingTop: number = 0;

    public loading: boolean = false;
    public captured: boolean = false;
    public rejectedCamera: boolean = false;
    public container: HTMLDivElement = undefined;
    public canvas: HTMLCanvasElement = undefined;
    public mediaStreamTrack: MediaStreamTrack;

    public constraints: MediaStreamConstraints = null;
    
    public async mounted() {
        this.container = this.$refs['media-camera-container'] as HTMLDivElement;
        this.canvas = this.$refs['media-camera-canvas'] as HTMLCanvasElement;
        CameraPreview.start({position: 'rear', parent: 'cameraPreview', className: 'cameraPreview'});
    }

    public async capturePicture() {
    }

    private async drawPictureOnCanvas(picture: string): Promise<void> {
        return new Promise((resolve,reject) => {
            try {
                let image = new Image();
                
                image.onload = async () => {
                    this.canvas.width = document.getElementsByClassName('camera-cordova')[0].clientWidth >= 480 ?  document.getElementsByClassName('camera-cordova')[0].clientWidth : 480;
                    this.canvas.height = document.getElementsByClassName('camera-cordova')[0].clientHeight >= 640 ? document.getElementsByClassName('camera-cordova')[0].clientHeight : 640;
                    var context = this.canvas.getContext('2d');
                    context.drawImage(image, 0, 0, this.canvas.width, this.canvas.height);
                    resolve();
                };
                image.src = `data:image/png;base64,${picture}`;
            } catch (error) {
                reject(error);
            }
        })
    }

    private async validatePicture() {
        setImmediate(async () => {
            const validations = [...this.validations];
            const errors: string[] = [];
            while (validations.length) {
                const validation = validations.shift();
                await validation(this.canvas, this.maskWidth, this.maskHeight).catch(error => errors.push(error));
            }

            this.unsetCameraMode();
            if (errors.length) {
                this.cancelPicture();
                this.$emit('on-capture-fail', errors);
            } else {
                this.captured = true;
            }

            this.loading = false;
        });
    }

    public cancelPicture() {
        // this.cameraPreview.stopCamera();
        // this.initializeCamera();
        // this.clearCanvasImage();
    }

    public confirmPicture() {
        // this.cameraPreview.stopCamera();
        // this.$emit('on-capture-confirm', this.canvas.toDataURL());
    }

    public onCloseModal() {
        this.rejectedCamera = false;
        // this.initializeCamera();
    }

    private getCameraPreviewOpts() {
        let cameraPreviewOpts = {
            x: 0,
            y: 0,
            width: this.containerWidth,
            height: this.containerHeight,
            camera: this.facingMode == ('user') ? 'front' : 'back',
            toBack: true,
        };
        return cameraPreviewOpts;
    }

    private setCameraMode() {
        if (this.facingMode == 'user') {
            // FIX ME CORDOVA
            // if (device.platform === 'Android') {
            //     this.cameraPreview.setZoom(this.containerZoom);
            // }
        }
        let html = document.getElementsByTagName('html')[0];
        html.className += 'camera-cordova';
    }

    private unsetCameraMode() {
        let html = document.getElementsByTagName('html')[0];
        html.className = '';
    }

    private clearCanvasImage() {
        setTimeout(() => { //Workaround para não mostrar o cinza de fundo
            let context = this.canvas.getContext('2d');
            context.scale(-1, 1);
            context.clearRect(0, 0, this.canvas.width, this.canvas.height);
            context.clearRect(this.canvas.width * -1, 0, this.canvas.width, this.canvas.height);
        }, 1000);
    }
}
```

BrunoTCouto, 2019-10-08:
There is lots of unused code, we are migrating from cordova to capacitor

But the error ocurs in the mounted section

arielhernandezmusa, 2019-10-08:
I'm not an expert on Vue, but is `mounted` called only once?

BrunoTCouto, 2019-10-08:
Yes

elepetit, 2019-10-08:
can you skip parent and class in the start command ?

elepetit, 2019-10-08:
CameraPreview.start({position:rear})

elepetit, 2019-10-08:
I think you should not import Capacitor-Camera-Preview line 3

elepetit, 2019-10-08:
import {
  Plugins,
  Capacitor
} from "@capacitor/core";
const { CameraPreview} = Plugins;

BrunoTCouto, 2019-10-08:
Im going test skiping the parent and the class.

if i don't import Capacitor-Camera-Preview in line 3 the camera doesn't open in the browser as well

elepetit, 2019-10-08:
What about import with Plugins and Capacitor, like

import {
Plugins,
Capacitor,
Camera-preview
} from "@capacitor/core";

elepetit, 2019-10-08:
not expert in vue but seems like your import already starts the camera preview, and when you call mounted it makes twice

elepetit, 2019-10-08:
I’m on holidays, will try to help you next week

BrunoTCouto, 2019-10-08:
Ok ! I'm already testing this options

Any update i will write here !

Thanks!

BrunoTCouto, 2019-10-08:
Without importing i get the following error:

Browser:
uncaught exception: CameraPreview does not have web implementation.

iOS:
{message: "camera already started", errorMessage: ""}

BrunoTCouto, 2019-10-08:
Just printed the condition that rejects in swift and ```self.cameraController.captureSession?.isRunning``` is ```nil``` so it should not enter in the if clause right ?

```swift 
if(self.cameraController.captureSession?.isRunning ?? false) {
    call.reject("camera already started")
}
```

elepetit, 2019-10-08:
yes

BrunoTCouto, 2019-10-08:
Looks like the error "camera already started" was my fault... For getting the error log i was refreshing the webview in safari...
Now the camera is starting correctly but not displaying in the screen

BrunoTCouto, 2019-10-08:
[image]

elepetit, 2019-10-08:
you should consider your css

BrunoTCouto, 2019-10-08:
In ios, where is expected the code put the "video"element ?

elepetit, 2019-10-08:
in your page’s scss
ion-content { --background: transparent !important; }

elepetit, 2019-10-08:
your preview is in the back

arielhernandezmusa, 2019-10-09:
As @elepetit says, the preview on android and ios is on a view back of root view, so to view the camera preview you have to make the  background transparent !!!

BrunoTCouto, 2019-10-09:
Fixed ! As I migrated the project from cordova to Ionic I forgot to add the ```<ion-app/>```and ```<ion-content/>``` tags. After this everything started working! 
Thanks !! :)
[image]

closed 2019-10-09 by BrunoTCouto

elepetit, 2019-10-09:
you’re welcome!

ryaa, 2023-10-30:
> not expert in vue but seems like your import already starts the camera preview, and when you call mounted it makes twice

@elepetit My app also has 'camera already started' error happening intermittently. Some users experience this error periodically and this is logged into our system however I can not reproduce this problem on my iOS devices.
I have the below import in **app.module.ts** and wondering if it might cause any problems? (note that i use capacitor 5 and the latest version of this plugin 5.0.0)
`import '@capacitor-community/camera-preview';`
Specifically, why do you think that it can make the camera to start twice?

I noticed that this was required to support the plugin on the web platform for the capacitor 3 (see https://github.com/capacitor-community/camera-preview/tree/capacitor-3-support#web-quirks) however this seems no longer required for the capacitor 5 (see https://github.com/capacitor-community/camera-preview#extra-web-installation-steps)