capacitor-community/camera-preview #11: BUG (Front camera doesn't open/ rear does it instead)

Yoshimatsup, 2019-10-25:
[image]

this condition is the main issue, in some devices, it works and then opens the front camera when I use CameraPreview.start({ position: 'front' }), but in another devices, for example my personal smartphone ( Xiaomi MI A2 Lite ), it doesn't, the rear camera is started instead.
I changed both results of this ternary if to Camera.CameraInfo.CAMERA_FACING_FRONT as a 'semi-workaround' and then opens the front camera in my phone, but the position argument on the .start function is useless, I'd appreciate some help.

Some fixes that would also be great:
1* After you take the picture, when you show it on the device screen it has a minimum zoom.
2* On most android/chrome devices, the picture you show on screen gets mirrorred, I use scaleX(-1) ot fix it, but could be a nice 'feature'

Yoshimatsup, 2019-10-28:
I've been reading the code, looks like defaultCamera is set to "back" by a Fragment, but nothing listen the argument  to change to '"front" on the .start call

arielhernandezmusa, 2019-10-28:
Ok, I'll see that problem

arielhernandezmusa, 2019-10-28:
Can you please give me Android version?

arielhernandezmusa, 2019-10-31:
@Yoshimatsup , I just fix the problem regarding to start with front or rear camera, the other issues, please separate in different issues and close this please.

Yoshimatsup, 2019-11-01:
Thanks for your help, i'll separate the issues and close this one.
Thanks again!

closed 2019-11-01 by Yoshimatsup