capacitor-community/camera-preview #13: Cannot access on Android

scoleman2272, 2019-12-27:
When I try to add "add(CameraPreview.class);" to the main activity I get an unresolved symbol error.

I would think that adding an import would fix the issue but this fails:
   
    import com.ahm.capacitor.camera.preview;

Something is not getting added to the android folder

    this.init(savedInstanceState, new ArrayList<Class<? extends Plugin>>() {{
          // Additional plugins you've installed go here
          // Ex: add(TotallyAwesomePlugin.class);
          add(CameraPreview.class);
    }});

benbut, 2020-01-28:
> When I try to add "add(CameraPreview.class);" to the main activity I get an unresolved symbol error.
>
> I would think that adding an import would fix the issue but this fails:
>
> ```
> import [...]

@scoleman2272 you need to import com.ahm.capacitor.camera.preview.**CameraPreview**;

arielhernandezmusa, 2020-01-28:
@scoleman2272 , after you import 

com.ahm.capacitor.camera.preview.CameraPreview;

Works or not?

arielhernandezmusa, 2020-01-29:
@scoleman2272 , this is my MainActivity class importing CameraPreview and is woking

`package com.ahm.capacitor.camera.preview.demo.react;

import android.os.Bundle;

import com.ahm.capacitor.camera.preview.CameraPreview;
import com.getcapacitor.BridgeActivity;
import com.getcapacitor.Plugin;

import java.util.ArrayList;

public class MainActivity extends BridgeActivity {
  @Override
  public void onCreate(Bundle savedInstanceState) {
    super.onCreate(savedInstanceState);

    // Initializes the Bridge
    this.init(savedInstanceState, new ArrayList<Class<? extends Plugin>>() {{
      // Additional plugins you've installed go here
      // Ex: add(TotallyAwesomePlugin.class);
      add(CameraPreview.class);
    }});
  }
}`

abhibly, 2020-02-07:
@arielhernandezmusa Hello, I followed the steps you mentioned above in my ionic4(angular) project:
This is MainActivity file:
`
package com.transistorsoft.backgroundgeolocation.ionic2;

import android.os.Bundle;

import com.ahm.capacitor.camera.preview.CameraPreview;

import com.getcapacitor.BridgeActivity;
import com.getcapacitor.Plugin;

import java.util.ArrayList;

public class MainActivity extends BridgeActivity {
  @Override
  public void onCreate(Bundle savedInstanceState) {
    super.onCreate(savedInstanceState);

    // Initializes the Bridge
    this.init(savedInstanceState, new ArrayList<Class<? extends Plugin>>() {{
      // Additional plugins you've installed go here
      // Ex: add(TotallyAwesomePlugin.class);
      add(CameraPreview.class);
    }});
  }
}

`
When i try to build android apk, I am getting `error: package com.ahm.capacitor.camera.preview does not exist`.
I couldn't find any info about this error. Can you please help how to fix this issue.

arielhernandezmusa, 2020-02-07:
Plase can you run `npx cap sync`, and then try to build android project again

abhibly, 2020-02-07:
@arielhernandezmusa Thank you. It worked.

closed 2020-02-18 by arielhernandezmusa