capacitor-community/camera-preview #15: Flipped image on Android

abhibly, 2020-02-08:
Hello,
I am using your plugin in the Ionic4-android app. When I capture the picture, the base64 data is returned and when I display it, the image is rotated. This issue is occurring when I tested on Android device. It works fine on web app.

This is the version I am using:
`"capacitor-camera-preview": "^0.2.0"`

I am attaching the screenshots of issue:
[image]
[image]

benbut, 2020-02-16:
I have the same problem

closed 2020-02-18 by arielhernandezmusa

NauelG, 2024-02-07:
Do you solve de issue?

commit cdba6fe9 (2020-02-18): fix android flip image close #15