capacitor-community/camera-preview #17: Extra options

nielsvanharen, 2020-03-10:
Hello,  great copy from the cordova plugin, we are using it in our migrated capacitor app atm. My only concern for now is the lack of default options. It would be great to offer more options available like the original plugin. It would be a great help to have at least the height and weight properties, but the same would be even better :)

arielhernandezmusa, 2020-04-03:
yeah, I will try to copy all features :-), but if you can help me, I'll feel  better

bialabs, 2020-04-07:
any advices on how to crop and resize the image from your plugin?

nielsvanharen, 2020-04-15:
> in the meantime you add all the extra options, any advices on how to crop and resize the image from your plugin?

   CameraPreview.capture(pictureOpts).then((base64PictureData) => {
                    // We pass width, height, x and y coordinates of our rectangle to crop func
                    // BEFORE crop function ends, it sends cropped base64 image to a server
                    this.crop(base64PictureData.value, rectWidth, rectHeight, xCoord, yCoord, (croppedBase64Image) => {
 //do stuff with your image
})

   crop(base64PictureData, rectWidth, rectHeight, xCoord, yCoord, callback: (param) => void) {

        // image will contain ORIGINAL image
        const image = new window.Image();

        // image will contain CROPPED image
        const canvas = document.createElement('canvas');
        const ctx = canvas.getContext('2d');

        // Load original image into image element
        image.src = 'data:image/png;base64,' + base64PictureData;
        image.onload = () => {

            const scaleHeight = rectHeight / image.height;
            const scaleWidth = rectWidth / image.width;

            let scale;
            let xImgUnits;
            let yImgUnits;
            if (scaleHeight < scaleWidth) {
                scale = scaleWidth;
                xImgUnits = 0;
                yImgUnits = (image.height - (rectHeight / scale)) / 2;
            } else {
                scale = scaleHeight;
                xImgUnits = (image.width - (rectWidth / scale)) / 2;
                yImgUnits = 0;
            }

            const rectWidthImgUnits = rectWidth / scale;
            const rectHeightImageUnits = rectHeight / scale;

            // Set canvas size equivalent to interpolated rectangle size
            canvas.width = rectWidthImgUnits;
            canvas.height = rectHeightImageUnits;

            ctx.drawImage(image,
                xImgUnits, yImgUnits,                       // Start CROPPING from (x_img_units, y_img_units).
                rectWidthImgUnits, rectHeightImageUnits,    // Crop interpolated rectangle
                0, 0,                                           // Place the result at 0, 0 in the canvas,
                rectWidthImgUnits, rectHeightImageUnits);   // Crop interpolated rectangle

            // Get base64 representation of cropped image
            const croppedBase64Image = canvas.toDataURL();

            // Now we are ready to send cropped image TO SERVER
            callback(croppedBase64Image);

            return croppedBase64Image;
        };
    }

this draws a box to take the picture from (license plate for example)

nielsvanharen, 2020-04-15:
the markup is terrible, stupid code plugin

nielsvanharen, 2020-05-01:
> yeah, I will try to copy all features :-), but if you can help me, I'll feel better

hey can you add me as a contributor? Ik have made some changes for the ios implementation on orientation. I can make a pr if you want, thanks

arielhernandezmusa, 2020-05-01:
I just added you

closed 2020-05-02 by arielhernandezmusa