capacitor-community/camera-preview #20: Documentation is poor

sourcer18, 2020-04-24:
the documentation of this plugin makes it practically impossible to know how to use it, I need some more documentation to know how to use it. Do you have any demo ?, I'm mounting it in ionic with angular and capacitor

nielsvanharen, 2020-05-01:
the main thing missing in the documentation is the html element that you need to provide.

`  CameraPreview.start({parent: 'cameraPreview', className: 'cameraPreview'})`

in my html it looks like this
```
<ion-content class="bg-transparent" no-padding>
    <div class="overlay" [ngClass]="'license'">
        <div id="cameraPreview" class="cameraPreview" #cameraPreview>
            <img class="take_pic_class" #take_pic_class src="assets/preview/btn_icon_mini.png">
        </div>
    </div>
</ion-content>
```
in the top of your page

```
    @ViewChild('cameraPreview', {static: true}) element: ElementRef;
    @ViewChild('take_pic_class', {static: true}) takePicBtnElement: ElementRef;
```
if the start of your CameraPreview.start or in the promise 

```
 const takePicBtn: any = this.takePicBtnElement.nativeElement;
 takePicBtn.onclick = () => {
                // Get rectangle coordinates
                const xCoord = this.rectCoords.left, yCoord = this.rectCoords.top;
                // Get rect width and height
                const rectWidth = rect.offsetWidth, rectHeight = rect.offsetHeight;
               //pictureOpts not used yet in plugin
                const pictureOpts = {
                    width: 640,
                    height: 480,
                    quality: 100
                };
                CameraPreview.capture(pictureOpts).then((base64PictureData) => {
                    //do something with your base64 string
                }, (err) => {
                    console.log(err);
                    this.loading.dismiss();
                    this.stopCamera();
                });
            };

```

arielhernandezmusa, 2020-05-31:
Yeah, I'll improve it, thanks.

closed 2020-05-31 by arielhernandezmusa