capacitor-community/camera-preview #21: capture() in dataURL format

suenot, 2020-04-25:
How to get data in dataURL format?

nielsvanharen, 2020-05-01:
the response is a base64 image format, you can use this as an image source or convert it to a blob if you want to save it as a file.

suenot, 2020-05-01:
Yes, I solved my problem. I converted base64 to dataUrl format and then converted to a blob format.
```js
// https://stackoverflow.com/a/38935990
function dataURLtoFile(dataurl, filename) {
  const arr = dataurl.split(',');
  // $flowignore
  const mime = arr[0].match(/:(.*?);/)[1];
  const bstr = atob(arr[1]);
  let n = bstr.length;
  const u8arr = new Uint8Array(n);
  while (n--) {
    u8arr[n] = bstr.charCodeAt(n);
  }
  return new File([u8arr], filename, {type:mime});
}

const base64PictureData = result.value;
const dataUrlPictureData = `data:text/plain;base64,${base64PictureData}`;
const f = dataURLtoFile(dataUrlPictureData, 'photo.jpeg');
```

closed 2020-05-01 by suenot