capacitor-community/camera-preview #27: Android - Live camera preview and capture have a gap

albanlorillard, 2020-05-09:
Hello,

Actually, i've 2 issues concerning capture() function and the live camera :

- On a first android mobile (Asus x00td / Android 8.1.0) that have only one back Camera, their is a gap between the live camera preview and the base64 capture.
At left: screenshot of my application
At right: the base64 image captured

[image] [image]

- On the second android mobile with 2 back camera (Samsung A70 / Android 9), the live camera preview is the first back camera, and capture function is shot with the 2nd back camera.
At left: screenshot of my application
At right: the base64 image captured

[image] [image]

Here is my configuration:
 ```
 CameraPreview.start({ 
      parent: "cameraPreview", 
      position: "rear", 
      camera: "rear"
    });
```

I've tried to add x:0, y:0 or width=window.screen.width and height=window.screen.height but that change only on the web browser.

Regards,

Alban

nielsvanharen, 2020-05-09:
This is something I noticed as well, the ‘old’ camera implementation of android does not support multiple camera’s very well , there is a new camera implementation that handles multiple  camera’s much better

hodeware, 2021-11-09:
Same issue here. Is there any solution?

pbowyer, 2021-11-10:
Looking at the photos in the original post, I am wondering if this is the same problem as I saw on an iPhone.

On many phones the camera isn't the same aspect ratio as the screen. CameraPreview covers the phone screen with the picture, which means it often crops off the sides of the picture when in portrait mode.

When the photo is captured it's the full photo, not the bit which fitted the screen.

It took me a while to work this out, and I added some JS in my app that cropped the captured photo to the aspect ratio of the phone's screen. For me this solved the problem.

### How to see if this is the problem
1. Fix your phone firmly somewhere - we don't want it to move
2. Start CameraPreview running
3. Take a screenshot on the phone of what you see
4. Capture the photo in CameraPreview
5. In a graphics program, open the captured photo and screenshot. Overlay the screenshot on the captured photo, resizing it as needed. Are they the same perspective? Do they fit exactly (so the same camera & lens was used)?

yksolanki9, 2021-12-24:
Hey @pbowyer facing the same issue. Would like to know how you worked it out. Please share the approach if you dont mind. Thanks!

mxlle, 2025-01-10:
I'm not fully sure if it is the same issue, but I had very similar symptoms and I was able to solve them by using `window.innerWidth` and `window.innerHeight` for width and height in the start and capture method. 
(Difference regarding status bar / footer)

Thought I leave this here in case it helps anyone else