capacitor-community/camera-preview #28: Problem after update to 0.6.0

suenot, 2020-05-09:
```
/Users/suenot/work/shakeapp-gallery/repos/mobile-app/node_modules/capacitor-camera-preview/android/src/main/java/com/ahm/capacitor/camera/preview/CameraPreview.java:95: error: cannot access AppCompatActivity
                    FragmentManager fragmentManager = getActivity().getFragmentManager();
                                                                 ^
  class file for android.support.v7.app.AppCompatActivity not found
```

```npx cap sync``` I made.

nielsvanharen, 2020-05-09:
The app compat v7 library isn’t used anymore, it should be resolved with the app compat 1.2.0 library, can you tell me what capacitor version and android version you use? This give me some things to look into

suenot, 2020-05-10:
Android 9
"@capacitor/android": "1.5.2"
"@capacitor/core": "1.5.1"

Android Studio 3.6.2
Build #AI-192.7142.36.36.6308749, built on March 18, 2020
Runtime version: 1.8.0_212-release-1586-b4-5784211 x86_64
VM: OpenJDK 64-Bit Server VM by JetBrains s.r.o
macOS 10.15.4
GC: ParNew, ConcurrentMarkSweep
Memory: 2014M
Cores: 4
Registry: ide.new.welcome.screen.force=true
Non-Bundled Plugins:

nielsvanharen, 2020-05-10:
the problem is with the androidx compatibility. when using an androidx plugin, you can resolve it by doing these steps.

Add the following two flags to true in your gradle.properties file at ProjectFolder/android/gradle.properties

android.useAndroidX=true
android.enableJetifier=true

Execute
npm install --save-dev jetifier
npx jetify

then in your package.json
add
"postinstall": "jetifier"

when you do an npx copy and open android studio the problem should be resolved.

arielhernandezmusa, 2020-05-31:
@suenot, please let me know if you solve the problem

closed 2020-12-06 by arielhernandezmusa