capacitor-community/camera-preview #30: CameraPreview.flip() is not a function

ASomerN, 2020-05-24:
I have a working module for this, however, on calling flip() I get the error from android devices (not tested iOS). I'm on AndroidX.

Msg: ERROR TypeError: CameraPreview.flip is not a function.

"capacitor-camera-preview": "^0.6.0",
"@angular/core": "~9.1.9",
"@capacitor/core": "^2.1.0",

```
import { Component, OnInit, Output, EventEmitter } from '@angular/core';
import {
  Plugins
} from '@capacitor/core';
const { CameraPreview } = Plugins;

@Component({
  selector: 'app-camera-preview',
  templateUrl: './camera-preview.component.html',
  styleUrls: ['./camera-preview.component.scss'],
})
export class CameraPreviewComponent implements OnInit {
  @Output() takePictureCamera = new EventEmitter<string>();

  constructor() { }

  ngOnInit() {
    CameraPreview.start({ 
      parent: "cameraPreview", 
      position: "front", 
      camera: "front"
    });
  }
  
  takePicture(){
      const pictureOpts = {
        correctOrientation: true,
        width: 150,
        quality: 50
      };
      CameraPreview.capture(pictureOpts).then((base64PictureData) => {
          this.takePictureCamera.emit('data:image/jpg;base64,'+base64PictureData.value);
          CameraPreview.stop();
      }, (err) => {
          console.log(err);
          CameraPreview.stop();
      });
  }

  flip(){
    CameraPreview.flip();
  }
}
```

arielhernandezmusa, 2020-05-24:
Did you add the plugin to MainActivity?

ASomerN, 2020-05-24:
Yes, but I've just spotted in the build that it throws: "error: cannot find symbol       add(CameraPreview.class);".

arielhernandezmusa, 2020-05-24:
Are you import CameraPreview?

ASomerN, 2020-05-25:
in Main Activity or in Angular? Adding anything else to MainActivity doesn't work.
It is imported in the project as it works (apart from flip() )
Where should I be adding additional import statement?

ASomerN, 2020-05-25:
Here's a repo: https://github.com/ASomerN/camera-preview-example
In this repo, I can't generate a .apk file if I have added the line for Android quirks: "add(CameraPreview.class);"
The emulator works with and without that line.

arielhernandezmusa, 2020-05-25:
You are missing import on main activity, please add

`import com.ahm.capacitor.camera.preview.CameraPreview;`

And let me know if works

ASomerN, 2020-05-26:
That doesn't resolve the issue. Still can't build a .apk file with those lines included in the MainActivity either.

arielhernandezmusa, 2020-05-31:
@ASomerN as I say, what you missing is importing CamperaPreview on MainActivity I just download you code and test and works !!!!

[image]

ASomerN, 2020-05-31:
I'm having no joy. I can get the camera working but not the flip.

I have added those lines (and I've updated the repo). I had the same issue in Android Studio where it wouldn't recognise the package. After reverting to an older Gradle version it allows the build to proceed but nothing works. 

I have then upgraded and now get: "android:failed: Cause: org/gradle/api/internal/java/usagecontext/LazyConfigurationUsageContext"
When building.

I'll need to overcome these issues with Android Studio which I assume is where the issues exist now.
Can you let me know what version of gradle you are using?

arielhernandezmusa, 2020-05-31:
I see you add

import com.getcapacitor.camera.preview.CameraPreview;

but the plugin is

import com.ahm.capacitor.camera.preview.CameraPreview;

also, grade 6.0+ not have support for LazyConfigurationUsageContext so, you need to use a lower grade version

arielhernandezmusa, 2020-05-31:
If you upgrade to capacitor 2.1.2, it will work with Gradle 6.1.1

arielhernandezmusa, 2020-05-31:
see this PR https://github.com/ASomerN/camera-preview-example/pull/1

ASomerN, 2020-06-01:
The commit with the "import com.getcapacitor.camera.preview.CameraPreview;" import was me trying anything to get it to work, I've pushed the change with the correct code I was using but as mentioned it still doesn't work.

Will upgrade capacitor and rebuild. It would be helpful to have a working example of camera.flip() being used! I can get the camera to function but not the preview and flip on native. The functionality difference between web and app is another issue.

ASomerN, 2020-06-01:
I've got it working but now there's no preview on initialising the module.
I'm closing this as the flip() issue is resolved but now there are different issues.

closed 2020-06-01 by ASomerN

SubhadeepJasu, 2021-02-27:
How did you fix the flip issue? @ASomerN