capacitor-community/camera-preview #31: Is there a working example?

v3rmin, 2020-06-02:
I can't get more than a blank page when following the README and deploying to an iOS device. Does anyone use this and have a working example to share?

nielsvanharen, 2020-06-02:
9/10 a blank page is caused bu not having as transparant background. you can check
https://github.com/arielhernandezmusa/capacitor-camera-preview/issues/20

v3rmin, 2020-06-02:
@nielsvanharen, your example was the one I was following (along with the README).

#### src/app/home/home.page.html
```html
<ion-content>

  <ion-content class="bg-transparent" no-padding>
    <div class="overlay" [ngClass]="'license'">
        <div id="cameraPreview" class="cameraPreview" #cameraPreview>
            <div class="shutter-button" #shutterButton></div>
        </div>
    </div>
  </ion-content>

</ion-content>

```

#### src/app/home/home.page.ts
```js
import { Component, ViewChild, ElementRef } from '@angular/core';
import { Plugins } from "@capacitor/core"

const { CameraPreview } = Plugins

@Component({
  selector: 'app-home',
  templateUrl: 'home.page.html',
  styleUrls: ['home.page.scss'],
})
export class HomePage {

  @ViewChild('cameraPreview', {static: true}) element: ElementRef;

  constructor() { }

  ngOnInit() {
    this.startCamera();
  }

  private startCamera() {
    CameraPreview.start({parent: 'cameraPreview', className: 'cameraPreview'})
  }

}
```

#### src/app/home/home.page.scss
```css
.overlay {
  position: absolute;
  width: 100%;
  height: 100%;
}

.cameraPreview {
  display: flex;
  width: 100%;
  height: 100%;
}

.shutter-button {
  background-size: 60px 60px;
  background-color: red;
  width: 55px;
  height: 55px;
  bottom: 20px;
  left: 0px;  right: 0px;
  position: absolute; /*it can be fixed too*/
  margin: auto;
  z-index:300000;
}

ion-content, ion-page, .nav-decor {
  background-color: transparent !important;
}

.bg-transparent {
  background-color: transparent !important;
}
```

--------

- capacitor-camera-preview (0.6.0)
- ionic (6.9.2)
- ios (13.5.1)
- xcode (11.5)

felixhhyeung, 2020-06-08:
[image]

@v3rmin even after I opened a blank new project and copy the code you provided, I cannot get the preview.

felixhhyeung, 2020-06-08:
Update: I can make this work by adding ```ion-content {--background: transparent!important;}``` to the page .scss.
Moreover, it will not work on PopupController (I think the preview is behind all ion-content with no regards on ```parent: 'cameraPreview'``` that set to the CameraPreview.start() for iOS app.).

closed 2020-12-07 by arielhernandezmusa