capacitor-community/camera-preview #32: Getting blank page with Ionic React

axcasella, 2020-06-04:
Hi, I am trying out this plugin but can only get a blank screen. Trying to piece together how this works without more instructions from the README is a bit difficult. Can someone point me to what I am missing? I have the following: 

App.js
```
import {
  IonApp,
  IonContent,
  IonHeader,
  IonToolbar,
  IonTitle
} from "@ionic/react";
import React, { useEffect } from "react";
import "./App.css";
import { Plugins } from "@capacitor/core";
const { CameraPreview } = Plugins;

const App = () => {
  useEffect(() => {
    startCamera();
  }, []);

  const startCamera = () => {
    CameraPreview.start({
      parent: "cameraPreview",
      position: "front",
      className: "cameraPreview"
    });
  };

  return (
    <IonApp>
      <IonHeader>
        <IonToolbar>
          <IonTitle>My App</IonTitle>
        </IonToolbar>
      </IonHeader>

      <IonContent className='ion-padding'>
        <div class='overlay'>
          <div id='cameraPreview' class='cameraPreview'></div>
        </div>
      </IonContent>
    </IonApp>
  );
};

export default App;
```

In App.css, I have made sure to set the entire **ion-content** to transparent

```
.overlay {
  position: absolute;
  width: 100%;
  height: 100%;
}

.cameraPreview {
  display: flex;
  width: 100%;
  height: 100%;
}

ion-content {
  background-color: transparent !important;
}

```

Upon inspecting the elements, I have verified that ion-content is indeed transparent.

damngamerz, 2020-08-27:
I'm also facing the same problem with Ionic React. Not sure How to solve this.

damngamerz, 2020-08-28:
Dived deep into the code. If I remove this line
https://github.com/capacitor-community/camera-preview/blob/81cacb920f6ab0da116d3d88ea1bb5e6a9ae37c3/android/src/main/java/com/ahm/capacitor/camera/preview/CameraPreview.java#L265
The camera shows up over the whole screen. :)
I really don't understand what this line does. But I have to dive deeper into Android developer api to understand it.

Btw is there a way to bind it to the div element by passing id to `parent` like we have it for a webview. I couldn't find any reference for it in android files.

If not it would be wise that the startCamera function accepts the origin arguments (x,y) as well. Just like cordova-camera-preview-plugin.

If anyone has further ideas. Feel free to ping me here.

commit dc6d49e2 (2020-08-31): Merge pull request #48 from gastromatic/ss/android-fix

fix: add origin (x,y) to startCamera Close #32 Close #47
closed 2020-08-31 by arielhernandezmusa