capacitor-community/camera-preview #35: Import CameraPreview class

PIXPOINT, 2020-07-07:
How do I import the CameraPreview class into the MainAcitify.java? I am getting an error:

error: cannot find symbol add(CameraPreview.class);

closed 2020-07-08 by PIXPOINT

koczka, 2020-07-08:
What was the solution?

PIXPOINT, 2020-07-08:
Add:
` import com.ahm.capacitor.camera.preview.CameraPreview;`

in MainAcitify.java

thenaim, 2020-10-18:
@PIXPOINT, not working.
`error: package com.ahm.capacitor.camera.preview does not exist import com.ahm.capacitor.camera.preview.CameraPreview;`

any idea?

PIXPOINT, 2020-10-18:
@thenaim i don't use this plugin anymore. I switched to this:
https://github.com/cordova-plugin-camera-preview/cordova-plugin-camera-preview

thenaim, 2020-10-18:
@arielhernandezmusa

bravoivang, 2021-03-02:
> @thenaim i don't use this plugin anymore. I switched to this:
> https://github.com/cordova-plugin-camera-preview/cordova-plugin-camera-preview

That plugin doesnt work well. You cant put html elements infront of the preview. There is a bug open related to that