capacitor-community/camera-preview #37: Plugin version 1.0.0 gives an error when running capacitor sync command

ryaa, 2020-07-21:
**Describe the bug**
The below error when running 'npx cap sync' after the the latest plugin version 1.0.0 is installed
```
✖ Updating iOS native dependencies with "pod install" (may take several minutes):
✖ update ios:
[error] Error running update: Analyzing dependencies
[!] No podspec found for `CapacitorCommunityCameraPreview` in `../../node_modules/@capacitor-community/camera-preview`
```

**To Reproduce**
Steps to reproduce the behaviour:
Pre-requisites: installed capacitor and capacitor ios platform
1. Install the latest plugin version 1.0.0
2. Run npx cap sync
Note that the above error occurs

**Expected behavior**
No error should happen

**Additional context**
N/A

alenmestrov, 2020-07-21:
@ryaa I needed to change name of .podspec file in node_modules/capacitor-camera-preview to CapacitorCameraPreview.podspec and also in that file s.name to CapacitorCameraPreview for sync to pass. I don't know if that fixes it but sync passes after that.

ryaa, 2020-07-22:
> @ryaa I needed to change name of .podspec file in node_modules/capacitor-camera-preview to CapacitorCameraPreview.podspec and also in that file s.name to CapacitorCameraPreview for sync to pass. I don't know if that fixes it but sync passes after that.

This is exactly what my PR does - see https://github.com/capacitor-community/camera-preview/pull/38 and this should fix the problem

arielhernandezmusa, 2020-07-22:
PR #38 merged

a-uz, 2020-07-22:
@ryaa I just installed the latest version 1.0.1 from npm and it's still an issue, since `CapacitorCommunityCameraPreview.podspec` file is missing. Could you take a look?

ryaa, 2020-07-23:
> @ryaa I just installed the latest version 1.0.1 from npm and it's still an issue, since `CapacitorCommunityCameraPreview`

Confirming the problem. package.json files property referenced the previous file name so the new renamed file was not included into the published npm package.

This PR should fix it https://github.com/capacitor-community/camera-preview/issues/37#issuecomment-662388451

scottieslg, 2020-08-02:
I just installed this and got the exact same error.  I'm on v1.0.1.

ryaa, 2020-08-02:
> I just installed this and got the exact same error. I'm on v1.0.1.

The PR with the latest fix has not been accepted/merged/released yet

bot101, 2020-08-03:
Currently facing this same issue.

ryaa, 2020-08-04:
As a temporary workaround, until my PR is merged, after you install the plugin, please unzip and copy the below file into node_modules/@capacitor-community/camera-preview

[CapacitorCommunityCameraPreview.podspec.zip](https://github.com/capacitor-community/camera-preview/files/5020827/CapacitorCommunityCameraPreview.podspec.zip)

bot101, 2020-08-05:
Ok @ryaa, thanks.

arielhernandezmusa, 2020-08-05:
PR #39 merged.

closed 2020-08-05 by arielhernandezmusa