capacitor-community/camera-preview #40: Not working on android, iOS works fine

alenmestrov, 2020-07-23:
**Describe the bug**
When I want to take picture, I'm getting this error: 

E/AndroidRuntime: FATAL EXCEPTION: Thread-9
    Process: io.ionic.submarineLoyalty, PID: 31034
    java.lang.RuntimeException: takePicture failed
        at android.hardware.Camera.native_takePicture(Native Method)
        at android.hardware.Camera.takePicture(Camera.java:1611)
        at android.hardware.Camera.takePicture(Camera.java:1553)
        at com.ahm.capacitor.camera.preview.CameraActivity$6.run(CameraActivity.java:644)

After that application shuts down.
Also, I don't see camera preview, just black screen. To be precise, when I come to page, page is transparent (white), then, when I allow for application to take images it turns black. 

**Expected behavior**
Take image :)

**Screenshots**
Let me know if they can help understand the issue and I will deliver them immediately. 

**Smartphone (please complete the following information):**
 - Device: SM-J530F
 - OS: Android 9
 - Browser Samsung Internet, Google Chrome 
 - Version [e.g. 22]

**Additional context**

What I tried to do to resolve this issue:
uninstalled plugin
installed plugin
ionic build
npx cap sync android
also I removed and added again android platform

What I noticed is that when I run npx cap sync android I'm getting this info:
✔ Copying web assets from www to android/app/src/main/assets/public in 689.05ms
✔ Copying native bridge in 1.43ms
✔ Copying capacitor.config.json in 1.08ms
  Found 4 Cordova plugins for android
    cordova-plugin-advanced-http (2.4.1)
    cordova-plugin-device-motion (2.0.1)
    cordova-plugin-file (6.0.2)
    cordova-plugin-shake (0.7.0)
✔ copy in 958.97ms
✔ Updating Android plugins in 28.34ms
  Found 1 Capacitor plugin for android:
    **@capacitor-community/camera-preview (1.0.0)**
  Found 4 Cordova plugins for android
    cordova-plugin-advanced-http (2.4.1)
    cordova-plugin-device-motion (2.0.1)
    cordova-plugin-file (6.0.2)
    cordova-plugin-shake (0.7.0)
✔ update android in 127.55ms
Sync finished in 1.098s

but when I run: ionic capacitor run android - I'm getting this info:
✔ Copying web assets from www to android/app/src/main/assets/public in 3.00s
✔ Copying native bridge in 3.45ms
✔ Copying capacitor.config.json in 4.37ms
⠋ copy  Found 4 Cordova plugins for android
    cordova-plugin-advanced-http (2.4.1)
    cordova-plugin-device-motion (2.0.1)
    cordova-plugin-file (6.0.2)
    cordova-plugin-shake (0.7.0)
✔ copy in 3.26s

In the second part @capacitor-community/camera-preview (1.0.0) isn't shown. I don't know if that has anything to do with the problem but its just an FYI

Full debug in android studio looks like this:

D/Preview: screen is rotated 0deg from natural
D/Preview: back camera is oriented -90deg from natural
D/Preview: need to rotate preview 90deg
I/System.out: camera started
D/CameraActivity: cameraCurrentlyLocked:0
D/Preview: optimal preview size: w: 176 h: 144
D/Preview: optimal preview size: w: 176 h: 144
D/ViewRootImpl@2818e74[MainActivity]: Relayout returned: old=[0,0][720,1280] new=[0,0][720,1280] result=0x1 surface={true 3578068992} changed=false
D/Preview: optimal preview size: w: 176 h: 144
D/Preview: optimal preview size: w: 176 h: 144
D/Preview: optimal preview size: w: 720 h: 720
V/Capacitor/Network: Notifying listeners for event networkStatusChange
D/Capacitor/Network: No listeners found for event networkStatusChange
D/Preview: optimal preview size: w: 176 h: 144
D/ViewRootImpl@2818e74[MainActivity]: ViewPostIme pointer 0
D/ViewRootImpl@2818e74[MainActivity]: ViewPostIme pointer 1
I/Capacitor/Console: File: http://localhost/camera-preview-camera-preview-module-es2015.js - Line 159 - Msg: take picture
V/Capacitor/Plugin: To native (Capacitor plugin): callbackId: 113432080, pluginId: CameraPreview, methodName: capture
V/Capacitor: callback: 113432080, pluginId: CameraPreview, methodName: capture, methodData: {"quality":100}
D/CameraActivity: CameraPreview takePicture width: 0, height: 0, quality: 100
D/CameraActivity: CameraPreview previewAspectRatio 1.7777777777777777
    CameraPreview optimalPictureSize 1920x1080
D/InputMethodManager: HSIFW - flag : 0 Pid : 31034
E/AndroidRuntime: FATAL EXCEPTION: Thread-9
    Process: io.ionic.submarineLoyalty, PID: 31034
    java.lang.RuntimeException: takePicture failed
        at android.hardware.Camera.native_takePicture(Native Method)
        at android.hardware.Camera.takePicture(Camera.java:1611)
        at android.hardware.Camera.takePicture(Camera.java:1553)
        at com.ahm.capacitor.camera.preview.CameraActivity$6.run(CameraActivity.java:644)
I/Process: Sending signal. PID: 31034 SIG: 9

Everything works on iOS - camera preview, flash functionality, take picture....

Do you maybe know what could be the problem here?

Thank you in advance,
Alen

alenmestrov, 2020-08-09:
I found a source of the problem... 
When initializing camera I didn't set width and height properties:
const cameraPreviewOptions: CameraPreviewOptions = {
  position: 'rear'
};
That is why the preview wasn't shown and when I wanted to take a picture it failed because it tried took an image with 0 width and 0 height... 
After setting width and height values to device width and height everything works.

sertal70, 2020-08-24:
Hi @alenmestrov, I'm struggling to make plugin work on iOS (camera preview doesn't show), would you be so kind to share a sample code of your implementation?

JyTosTT, 2020-09-24:
Hello @sertal70, I have created a [repository](https://github.com/JyTosTT/ionic-camera-preview) on what I found on this [issue](https://github.com/capacitor-community/camera-preview/issues/31#issuecomment-637362221). It works on Ios but not yet on Android.

I'm looking for Android now,
Regards

sertal70, 2020-09-25:
Thank you @JyTosTT , very appreciated!

flaupi, 2020-09-29:
@JyTosTT, thank you for sharing!

I have the same problem as @sertal70 and can't see the live image of the camera, while it actually takes photos. I can preview them and work with them. I will have a look on your project to see, what could be the issue with Android. There, everything works fine for me.

flaupi, 2020-09-29:
At first glance, only difference I can see is, that the `toBack` option is missing in the `CameraPreviewOptions` object. At least in some version of the camera preview package, this caused some problems for me.

Did you try connecting to the web console for further information? You can start the apk on your phone, connected to you pc. In chromium or so, you get access to your device on `chrome://inspect`.

JyTosTT, 2020-09-29:
Hello @flaupi,

Thank you for your feedback, I just tested and it works. I just don't have the preview displayed, I can only see the result but it's already good to know! I will watch it later, thank you :)

flaupi, 2020-09-30:
@JyTosTT, on which platform are you testing?

* I see a live camera picture and a resulting photograph on Android with my implementation
* I see a black screen, but a resulting photograph on iOS with my project
* _I actually see a live picture and a resulting photograph on Android with your sample project_
* I see a black screen, but a resulting photograph on iOS with your sample project

JyTosTT, 2020-10-01:
Hello @flaupi, I tested on Android.

For my project :
On Ios, I see the camera preview and the resulting photograph.
On Android, I see a black screen but the resulting photograph.

flaupi, 2020-10-01:
Hi @JyTosTT,

with your original project or with the demo project as well?

JyTosTT, 2020-10-01:
No, it's on my [demo project](https://github.com/JyTosTT/ionic-camera-preview-demo)!

flaupi, 2020-10-01:
I'm really confused, seeing it actually platforms switched.

For Android, it should be easy to connect developer tools to your app on the phone. Can you see some `z-index` stuff probably being messed up in the styling?

Or for Android, you should be able to test the camera in the emulator. Could you see a different behaviour in the emulator?

rcluan, 2020-10-05:
I'm having some trouble here with Android. Camera starts fine, I can see the preview and the html correctly, but when I use the capture method I've got the following in Android Studios' logcat:

```
2020-10-05 18:29:45.876 13256-13382/<package> V/Capacitor: callback: 117621990, pluginId: CameraPreview, methodName: capture, methodData: {"quality":100}
2020-10-05 18:29:45.877 13256-13357/<package> D/CameraActivity: CameraPreview takePicture width: 0, height: 0, quality: 100
2020-10-05 18:29:45.880 13256-13527/<package> D/CameraActivity: CameraPreview previewAspectRatio 1.3333333333333333
2020-10-05 18:29:45.881 13256-13527/<package> D/CameraActivity: CameraPreview optimalPictureSize 1280x960
```

Have you guys ran into this before? I don't understand why height and width are zeroed as they are set when I start the camera.

JyTosTT, 2020-10-06:
Hello @flaupi,

For the emulator, it works but after a few images, the application crashes with:
`E/AndroidRuntime: FATAL EXCEPTION: main
    Process: io.ionic.test.camera, PID: 9842
    java.lang.RuntimeException: Fail to connect to camera service
        at android.hardware.Camera.<init>(Camera.java:551)
        at android.hardware.Camera.open(Camera.java:396)
        at com.ahm.capacitor.camera.preview.CameraActivity.onResume(CameraActivity.java:258)
        at android.app.Fragment.performResume(Fragment.java:2580)
        at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:1346)
        at android.app.FragmentManagerImpl.moveFragmentToExpectedState(FragmentManager.java:1581)
        at android.app.FragmentManagerImpl.moveToState(FragmentManager.java:1642)
        at android.app.FragmentManagerImpl.executeOpsTogether(FragmentManager.java:2220)
        at android.app.FragmentManagerImpl.removeRedundantOperationsAndExecute(FragmentManager.java:2166)
        at android.app.FragmentManagerImpl.execPendingActions(FragmentManager.java:2067)
        at android.app.FragmentManagerImpl$1.run(FragmentManager.java:742)
        at android.os.Handler.handleCallback(Handler.java:883)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:214)
        at android.app.ActivityThread.main(ActivityThread.java:7356)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:930)`

But I have the preview, not like I test the apk on my android! 
I don't see any particular error suddenly with the layers `z-index`

Regards

flaupi, 2020-10-12:
@JyTosTT, in the emulator, I could get rid of this error by changing the camera configuration of the virtual device. By default, the front camera was set to `emulated` and the back camera was set to `VirtualScene`. I kept the back camera as is, but switched the front camera to `Webcam0` and it started to work in the emulated device.

ruanitto, 2020-12-06:
> I'm having some trouble here with Android. Camera starts fine, I can see the preview and the html correctly, but when I use the capture method I've got the following in Android Studios' logcat:
> 
> ```
> 2020-10-05 18:29:45.876 13256-13382/<package> V/Capacitor: callback: 117621990, pluginId: CameraPreview, methodName: capture, methodData: {"quality":100}
> 2020-10-05 18:29:45.877 13256-13357/<package> D/CameraActivity: CameraPreview takePicture width: 0, height: 0, quality: 100
> 2020-10-05 18:29:45.880 13256-13527/<package> D/CameraActivity: CameraPreview previewAspectRatio 1.3333333333333333
> 2020-10-05 18:29:45.881 13256-13527/<package> D/CameraActivity: CameraPreview optimalPictureSize 1280x960
> ```
> 
> Have you guys ran into this before? I don't understand why height and width are zeroed as they are set when I start the camera.
> 
> EDIT: There is also an error log
> 
> ```
> E/Camera: Error 2
> E/Camera: Error 1
> ```

Same error for me

ruanitto, 2020-12-06:
I made an pull request for this issue:

https://github.com/capacitor-community/camera-preview/pull/73

closed 2020-12-06 by arielhernandezmusa