capacitor-community/camera-preview #43: Camera preview option "position" doesn't seem to affect the preview

bcrespy, 2020-08-05:
**Describe the bug**
The front camera preview is working, however the option "position" set to "rear" does not change the position of the activated camera.

`import { Plugins, CameraDirection } from "@capacitor/core"
const { CameraPreview } = Plugins;
import { CameraPreviewOptions } from '@capacitor-community/camera-preview';

const cameraPreviewOptions: CameraPreviewOptions = {
  parent: 'test-cam',
  position: 'rear',
  height: 1920,
  width: 1080
};

@Component({ ... })
export class Content360Component implements OnInit {
  ngOnInit() {
    CameraPreview.start(cameraPreviewOptions);
  }
}
`

**Expected behavior**
The rear camera preview should be initialized instead of the front.

**Smartphone (please complete the following information):**
 - Device: Android
 - OS: 8.1
 - Browser: WebView native App, built with Ionic 6.10.2

**Additional context**

**This is the first mobile app I develop, please tell me if you need more informations regarding that particular issue.**

On a side note, I wasn't able to add the plugin to the MainActivity file, as stated in the readme:
`this.init(savedInstanceState, new ArrayList<Class<? extends Plugin>>() {{
      // Additional plugins you've installed go here
      // Ex: add(TotallyAwesomePlugin.class);
      add(CameraPreview.class);
}});`
CameraPreview.class cannot be found when building, I suspect I am missing an import but I couldn't find what I am supposed to import.

I tried:

`import com.ahm.capacitor.camera.preview;
import com.getcapacitor.CameraPreview`

But none succeeded.

flaupi, 2020-08-10:
I can confirm this issue and add a few details:

I could not add `CameraPreview.class` as well.

I added this function for better readability:

```
private startCamera() {
  CameraPreview.start({
    position: 'rear',
    parent: 'cameraPreview',
    className: 'cameraPreview',
    height: 1920,
    width: 1080,
    rotateWhenOrientationChanged: false,
  });
}
```

I also tried to add a `CameraPreviewOptions` object, but than, things weren't compiling properly, because `rotateWhenOrientationChanged` allegedly doesn't exist. Weird thing is, the compiler doesn't complain when I add the option like I do in my function. But anyways, there is no effect.

I also tried to add a button to switch the camera by hand. Documentation says that it is `CameraPreview.flip()`, but this function doesn't exist.

In addition: I tried different values for `height`, `width` and `rotateWhenOrientationChanged` and nothing seems to have an effect either.

The camera-preview version on my machine is the tag 1.0.2 and I'm using ionic with angular in the newest releases.

flaupi, 2020-08-31:
Update:

I had some trouble, adding the plugin within the androind `MainActivity`. After a few updates and experiments, AndroidStudio was able to suggest and actually resolve

`import com.ahm.capacitor.camera.preview;`

After this, I was able to start the camera with the correct camera and so on.

dmitryVashchenko, 2020-09-26:
> Update:
> 
> I had some trouble, adding the plugin within the androind `MainActivity`. After a few updates and experiments, AndroidStudio was able to suggest and actually resolve
> 
> `import com.ahm.capacitor.camera.preview;`
> 
> After this, I was able to start the camera with the correct camera and so on.

Hello, tell me please how you fix this issue?

flaupi, 2020-09-30:
@dmitryVashchenko, I honestly don't know :(

I think, it was a series of several tries with `npm ci` or `npm install`, `ionic buid` and `npx cap sync`. It just started working after several tries.

flaupi, 2020-10-14:
As far, as I understand:

If the Plugin is not added in the `MainAcitivity`, the Plugin is just not working at all, but the smartphone defaults to the web implementation. And it seems that the rear camera is just not implemented for the web platform.

johnbyrne7, 2021-02-15:
I'm also having this same problem.  My MainActivity looks OK, as far as I can tell:

`import com.ahm.capacitor.camera.preview.CameraPreview;

public class MainActivity extends BridgeActivity {
  @Override
  public void onCreate(Bundle savedInstanceState) {
    super.onCreate(savedInstanceState);

    // Initializes the Bridge
    this.init(savedInstanceState, new ArrayList<Class<? extends Plugin>>() {{
      // Additional plugins you've installed go here
      // Ex: add(TotallyAwesomePlugin.class);
      add(CameraPreview.class);
    }});
  }
}

However when I run on Android, the rear camera does not work, and I get this error on flip() attempt:

    Error: flip not supported under the web platform

JB

SubhadeepJasu, 2021-02-27:
Is still doesn't work. I can use the web  cam when in browser and only the front camera when using my phone. The button just doesn't work.

PR #105 (2021-03-22): Web rear facing camera
Consuming `options.position` in web portion of the plugin so rear camera can be accessed on mobile devices.

Using [facingMode: environment](https://developer.mozilla.org/en-US/docs/Web/API/MediaTrackConstraints/facingMode), which will select an environment facing camera if there is one. Desktop still resorts to webcam.

Could fix #43 , I can't tell if the issue is directly related to web, android, or both?
closed 2021-03-22 by arielhernandezmusa