capacitor-community/camera-preview #49: Camera is always on top on Android after v1.0.3 update

sertal70, 2020-09-01:
After v1.0.3 update, camera preview shows always on top of any html elements on Android. It seems that the root cause is the deletion of [row 265](https://github.com/capacitor-community/camera-preview/blob/75a7ac471af24e4138e421cf19313d8d98f53dfe/android/src/main/java/com/ahm/capacitor/camera/preview/CameraPreview.java#L265) in `CameraPreview.java`. In fact, adding that line again restores the correct camera preview z-placement on the web view. 

@damngamerz you are the author of the PR where that line has been deleted, can you have a look at this?

ryaa, 2020-09-01:
I don't think that this line 265 should be deleted. Deleting smth just because you don't understand what it is is a very bad idea (see https://github.com/capacitor-community/camera-preview/pull/48#issue-476367352)

damngamerz, 2020-09-01:
Okay now I get it. That's why I proposed the changes. So someone could point it out at PR level. Anyway, I need this at very top level of html. Would it makes sense to have a boolean setting to have the preview on the front or back?
@sertal70 @ryaa

ryaa, 2020-09-01:
> Would it makes sense to have a boolean setting to have the preview on the front or back?

See toBack prop in cordova-plugin-camera-preview plugin https://github.com/cordova-plugin-camera-preview/cordova-plugin-camera-preview#methods

sertal70, 2020-09-01:
@damngamerz I hadn't enough time to test your PR before it were merged, sorry for that. As a heavy former user of cordova-plugin-camera-preview I would welcome a `toback` setting as suggested by @ryaa,

PR #50 (2020-09-03): fix(android): add toBack boolean to startCamera
Fixes #49 
I haven't tested this yet. But it should work. (I can only test this tomorrow)
closed 2020-09-03 by arielhernandezmusa