capacitor-community/camera-preview #53: iOS Camera Output not Visible

scanlop-pses, 2020-09-18:
**Describe the bug**
Using iOS 13.7 the camera output appears to be a white screen. It seems to be hidden behind this. A picture can still be taken however the preview is not visible. No errors are being generated.

**To Reproduce**
Steps to reproduce the behavior:
1.  Build Project
2. Install on Device
3. Open Camera
4. White screen
5. Take Picture
6. No Errors

**Expected behavior**
Camera preview should appear. 

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Smartphone (please complete the following information):**
 - Device: iPhone 11
 - OS: iOS 13.7

**Additional context**
No issues reported on Android.

Background is set to transparent.
`
ion-content {
  --background: transparent!important
}
`

flaupi, 2020-09-29:
I see the same behaviour on an iPhone 8 with iOS 13.51.

davehearne, 2020-09-29:
I have the same issue with iOS 13.7 on iPhone X
Working on Android

flaupi, 2020-09-29:
My implementation also works on Android :thinking:

closed 2020-12-07 by arielhernandezmusa

sarahizabel7, 2021-08-23:
I have the same issue on Iphone 11. Even after the merged fix https://github.com/capacitor-community/camera-preview/pull/69

If I use the `toBack : true` option, the camera preview don't show up.
It only shows if I use `toBack : false`.

Can anyone help?