capacitor-community/camera-preview #54: Build fails on Xcode 12 for iOS 13/14

sertal70, 2020-09-24:
**Describe the bug**
Build on Xcode 12 fails when targeting simulator with iOS 13.5 or 14 with the error:

`Value of type 'AVCapturePhotoOutput' has no member 'supportedFlashModes'`

on source `CameraController.swift` at line 228.

Strange thing is that if I target a real device with iOS 13.5 the build process completes without any error.

**Desktop (please complete the following information):**
 - OS: macOS 10.15.6

**Smartphone (please complete the following information):**
 - Device: iPhone 7 Plus
 - OS: 15.5.1

damngamerz, 2020-09-29:
https://github.com/Yummypets/YPImagePicker/issues/553
Well known problem with Xcode 12. There's a workaround. But I don't really understand swift code. Can anybody experienced take look?

damngamerz, 2020-09-30:
Concrete info https://developer.apple.com/forums/thread/86810

closed 2021-01-15 by arielhernandezmusa