capacitor-community/camera-preview #59: Is there a problem with higher resolutions?

flaupi, 2020-10-12:
**Is your feature request related to a problem? Please describe.**
When I'm photographing, the results are always in a very low resolution.

**Describe the solution you'd like**
The automatic search for a matching resolution should not default to very low resolutions.

**Describe alternatives you've considered**
I didn't really consider alternatives, because I just commented out the

```java
if ((width != 0 && height != 0) || (supportedSize.width * supportedSize.height < 2048 * 1024))
```

and got better results and no problems.

**Additional context**
There is some kind of a hard limit on resolutions (at least in the code for Android, I didn't look into the code for iOS) and it looks pretty arbitrary to me. Is there a solid reason for this limit? Is this just some tech debt from earlier times of the development?

anth0, 2021-01-24:
Can somebody from the project answer this question please? I am also puzzled as to why there would be a limitation do 2048 x 1024. The "fix" is simple on Android as we just need to comment out the lines mentioned by flaupi but what about on iOS?

anth0, 2021-01-26:
@flaupi I've discovered that adding width and height parameters on the CameraPreviewPictureOptions helps for that issue. By adding something like _width: 5000 height:2500_ it successfully takes me the highest resolution my device has available.

AlvaroMartinezQ, 2025-01-14:
I managed to get the highest resolution of a picture in Android whit this:

```ts
const captureImage = async () => {
  const cameraPreviewPictureOptions: CameraPreviewPictureOptions = {
    quality: 100,
  };
  
  // We use Quasar, you will need to somehow identify the platform you are running on or remove this if statement
  if ($q.platform.is.android) {
    cameraPreviewPictureOptions.width = window.screen.height;
    cameraPreviewPictureOptions.height = window.screen.width;
  }
  try {
    CameraPreview.setFlashMode({ mode: 'off' });
    const result = await CameraPreview.capture(cameraPreviewPictureOptions);
    let base64PictureData = result.value;

    // Do something with the base64

    stopCamera(); // Stop camera after capturing the image
  } catch (error) {
    console.error('Error capturing image:', error);
  }
}
```

Unfortunately it blocks the screen for some seconds before capturing the image.