capacitor-community/camera-preview #65: Prevent plugin to use audio permission

jikseyres16, 2020-10-29:
**Describe the bug**
Don't know if a bug or not,
I just want the camera functionality not the video, but on android it keeps informing to allow audio permission.

**Expected behavior**
Just do not want to show the audio permission on first install.

**Desktop (please complete the following information):**
 - OS: Android
 - Version 10

**Smartphone (please complete the following information):**
 - Device: Samsung A50
 - OS: Android
 - Browser stock browser
 - Version 10

arielhernandezmusa, 2020-12-06:
I think we can remove `Manifest.permission.RECORD_AUDIO` from `CameraPreview.java` to prevent audio permission request

jikseyres16, 2020-12-09:
Thanks, that did the trick. 👍

ryaa, 2021-02-22:
Manifest.permission.RECORD_AUDIO still has not been removed from CameraPreview.java and installed the plugin version 1.2.0 causes the problem as described in this ticket.
Here is the PR to fix this https://github.com/capacitor-community/camera-preview/pull/100

vkyeswa, 2021-05-20:
In v1.2.1 the  Manifest.permission.RECORD_AUDIO is completely removed thus not being requested causing the plugin to fail when recording video (Android)

ryaa, 2021-05-21:
> In v1.2.1 the Manifest.permission.RECORD_AUDIO is completely removed thus not being requested causing the plugin to fail when recording video (Android)

This permission should not be always requested when the plugin starts. This was the initial implementation and I think that this bug has been introduced when the video support has been added to this plugin so I reverted it back. My assumption that this RECORD_AUDIO permission should be requested only when the plugin is going to start for video recording.

ryaa, 2023-03-09:
It seems that this ticket can be closed as the required change has been merged and included into the latest release

jcesarmobile, 2023-06-08:
closing since the change was merged and released

closed 2023-06-08 by jcesarmobile