capacitor-community/camera-preview #67: flip method doesn't work

mwanginjamb, 2020-11-11:
**Describe the bug**

First of thanks for the effort. I am using this plugin in an ionic 4 project.

My problems are 2:

1. The position option does not work defaults to front even if I explicitly set it to rear.
2. The flip method is none responsive.

Question : are there special permissions I need to add to android manifest file?

Question 2: What remedy do you suggest because I need to use the rear camera?

ariari90, 2020-11-13:
I'm facing the same problem. Any update?

mwanginjamb, 2020-11-15:
> I'm facing the same problem. Any update?

Not really, am even thinking to switching to Cordova Version of this.

arielhernandezmusa, 2020-12-06:
@mwanginjamb flip function is working for me on iOS 14+

mwanginjamb, 2020-12-07:
> @mwanginjamb flip function is working for me on iOS 14+

Could you share the implementation?

arielhernandezmusa, 2020-12-09:
@mwanginjamb I upload an example with iOS working https://github.com/capacitor-community/camera-preview/tree/master/demo I test it on iPhone XS Max with iOS 14.2

sebaguse, 2021-01-09:
I had the same problem, however after adding 'parent' option with an element ID, it works fine.

closed 2021-01-15 by arielhernandezmusa