capacitor-community/camera-preview #68: Image size should be returned

mrSingh007, 2020-11-19:
As plugin not returning image size, we can not control uploaded images size to app. We want to restrict user from uploading big size images. Obviously there is 'quality' config attribute but it's not about image size. 

**Describe the solution you'd like**
Either on image response the image size should be returned aswell, so that on frontend size check can be done.
Or On config it should be possible to add max image size and plugin should check the image size.

The "exif" data can contain image size Information.

arielhernandezmusa, 2020-12-07:
Hi, @mrSingh007 did you refer to image file size? or image dimensions?

mrSingh007, 2020-12-07:
i mean image file size. in KB or MB

arielhernandezmusa, 2020-12-09:
You can calculate de image size from the base 64 result

mrSingh007, 2020-12-10:
yes, I can but then have to write the plugin again thats why I created this ticket for this feature.

arielhernandezmusa, 2020-12-10:
You don’t need to write the plug-in, take a look at this post https://softwareengineering.stackexchange.com/questions/288670/know-file-size-with-a-base64-string, you can calculate the file size  from the base64 returned by the plug-in. 

But  also I’ll add the feature you requested