capacitor-community/camera-preview #75: Android - Image base64 Rotation

ruanitto, 2020-12-09:
**Describe the bug**
Base64 of the image being rotated upside down on Android

**To Reproduce**
Capture image
**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Android
 - Browser: Not aplicable
 - Version: 1.1.1

**Smartphone (please complete the following information):**
 - Device: Xiaomi / Samsung / Motorola / AVD
 - OS: 8, 9, 10 (Versions tested)
 - Browser:  Not aplicable
 - Version: 1.1.1

**Additional context**

I performed the correction in the file CameraActivity.java and after performing a sequence of tests, everything went the right way.
I'm going to open a pull request.

arielhernandezmusa, 2020-12-09:
Okay, welcome PR :-)

closed 2020-12-09 by arielhernandezmusa

nncl, 2020-12-14:
Is this updated on the latest version? I'm asking that cause version 1.1.1 isn't updated, as you can see https://github.com/capacitor-community/camera-preview/blob/v1.1.1/android/src/main/java/com/ahm/capacitor/camera/preview/CameraActivity.java#L663.

Am I doing something wrong?

Thanks for your time.

ruanitto, 2020-12-15:
> Is this updated on the latest version? I'm asking that cause version 1.1.1 isn't updated, as you can see [...]

283 / 5000
Resultados de tradução
No, version 1.1.1 does not have this update yet, for now I directly change the file in node_modules / @ capacitor-community / camera-preview / android / src / main / java / com / ahm / capacitor / camera / preview / CameraActivity.java

Overwriting the switch on line 662 with the content below:

```java
            switch (rotation) {
              case Surface.ROTATION_0:
                degrees = 270;
                // degrees = 0; // before change
                break;
              case Surface.ROTATION_90:
                degrees = 0;
                // degrees = 180; // before change
                break;
              case Surface.ROTATION_180:
                degrees = 90;
                // degrees = 270; // before change
                break;
              case Surface.ROTATION_270:
                degrees = 180;
                // degrees = 0; // before change
                break;
            }
```

commit fab4d014 (2020-12-09): Merge pull request #76 from ruanitto/master

Issue 75 - Image rotation Close #75