capacitor-community/camera-preview #81: Access the video stream from the camera on android

Berkelium0, 2020-12-23:
I'm making a face detection app using face.api on Ionic/Angular for android devices and I need to access the video stream for face.api functions to work. This works on on webview as camera preview creates a <video> html entity with an id which i can get with querySelector. However, This does not happen on android platforms.

When I check the html structure of the app with chrome debugger, the <video> object doesn't show up even though there is a video stream happening on the device from the camera, resulting in face.api not being able to retrieve the video information. To reiterate, this works well on web.

It would really help me if there was a way to reach the <video> object so i can detect faces with face.api. 
Thanks for the great plug-in btw!

IgorSamer, 2021-02-12:
I was testing this plugin for exactly the same purpose, to use it in conjunction with face-api.js.

I did the first tests on the web version and when I went to test on Android I forgot to add the plugin in `MainActivity`. It was then that the camera was started up normally, however, using the only implementation provided by me, which was the web version.

That way, the `<video>` tag was used and in your case, just like mine, you can access the video stream via querySelector :)

I hope it helped and good luck with the project!

jcesarmobile, 2023-06-08:
On native there is no video tag used but native code instead.
If you need a video tag, you don't really need the plugin, Capacitor supporst getUserMedia and using it in a video tag.