capacitor-community/camera-preview #83: Unable to use torch in preview

gbrits, 2021-01-06:
**Describe the bug**
```[error] - Unhandled Promise rejection: failed to set flash mode ; Zone: <root> ; Task: null ; Value: {"message":"failed to set flash mode","errorMessage":""} fromNative@user-script:2:224:23```

**To Reproduce**
```CameraPreview.setFlashMode({ flashMode: 'torch' });```

**Expected behavior**
Torch to turn on when previewing

**Desktop (please complete the following information):**
iOS

**Smartphone**
iPhone XS

gbrits, 2021-01-06:
Furthermore, trying to do it as the documentation describes:
```
const cameraPreviewFlashMode: CameraPreviewFlashMode = 'torch';
CameraPreview.setFlashMode(cameraPreviewFlashMode);
```
Just results in this error:
```
Argument of type '"torch"' is not assignable to parameter of type '{ flashMode: string; }'.
```

gbrits, 2021-01-06:
I also tried changing it to `{flashMode:'TORCH'}` just in case.
```
[error] - Unhandled Promise rejection: Flash Mode not supported ; Zone: <root> ; Task: null ; Value: {"message":"Flash Mode not supported","errorMessage":""} fromNative@user-script:2:224:23
```

seeka, 2021-02-19:
Hello, 
Same problem on Android, i can't use flash when camera is started in 'rear'.

pbowyer, 2021-12-16:
Sorry to see that you had problems with the flash mode. I'm closing this issue as the torch flash mode does work (I'm using it daily). The documentation does need improving as it took me a few days to figure out how to make it work reliably.

closed 2021-12-16 by pbowyer

ManoloAMD, 2023-01-25:
> Sorry to see that you had problems with the flash mode. I'm closing this issue as the torch flash mode does work (I'm using it daily). The documentation does need improving as it took me a few days to figure out how to make it work reliably.

Yeah @pbowyer  in doc:

setFlashMode(options)
`const CameraPreviewFlashMode: CameraPreviewFlashMode = 'torch';

CameraPreview.setFlashMode(cameraPreviewFlashMode);` is not working.

Solved:
`   const cameraPreviewFlashMode = { flashMode: 'off' };
    CameraPreview.setFlashMode(cameraPreviewFlashMode);` is working :)