capacitor-community/camera-preview #84: Unable to properly size the preview

jgrossrieder, 2021-01-07:
Hi,

I started using the camera-preview package. Most part is working fine.

I've this HTML:
```
<ion-header>
  <ion-toolbar color="primary">
    <ion-buttons slot="start">
      <ion-back-button defaultHref="spots"></ion-back-button>
    </ion-buttons>
    <ion-title>Checkin</ion-title>
  </ion-toolbar>
</ion-header>

<ion-content>
  <div class="overlay" [style.z-index]="cameraActive ? 10 : 0">
    <div id="cameraPreview" class="cameraPreview">
    </div>
  </div>
</ion-content>
```

This code to get the video:

```
  ionViewWillEnter() {
    const cameraPreviewOptions: CameraPreviewOptions = {
      position: 'rear',
      parent: 'cameraPreview',
      className: 'cameraPreview',
      width: 1920,
    };
    CameraPreview.start(cameraPreviewOptions);
    this.cameraActive = true;
  }

  ionViewWillLeave() {
    CameraPreview.stop();
    this.cameraActive = false;
  }

```

and this CSS:

```
ion-content {--background: transparent!important;}
 
.overlay {
  position: absolute;
  width: 100%;
  height: 100%;
  z-index: 10;
}
 
.cameraPreview {
  display: flex;
  width: 100%;
  height: 100%;
  position: absolute;
}

```

When I navigate to the page, the video doesn't use the full space. I used the chrome inspector, the  cameraPreview div is. It is taking 640x480px.  Either I'm missing something, either the dimensions are overridden?

Any idea? I'm using the version 1.1.3.

Thank you.

gbrits, 2021-01-10:
@jgrossrieder use window.screen.width for full. eg:
```
const cameraPreviewOptions: CameraPreviewOptions = {
      x: 0,
      y: 0,
      width: window.screen.width,
      height: window.screen.height,
      position: 'rear',
      parent: 'cameraPreview',
      className: 'previewer',
      toBack: true,
      rotateWhenOrientationChanged: true
};
```

gbrits, 2021-01-10:
Also, there's a problem with the orientation not working on landscape properly. I fixed it by placing this, after my CameraPreview.start call:

```
setTimeout(() => {
      window.screen.orientation.unlock();
      window.screen.orientation.lock('portrait');
      window.screen.orientation.unlock();
      window.screen.orientation.lock('landscape');
}, 500);
```

jgrossrieder, 2021-01-10:
> @jgrossrieder use window.screen.width for full. eg:
>
> ```
> const cameraPreviewOptions: CameraPreviewOptions = {
>       x: 0,
>       y: 0,
>       width: window.screen.width,
>       height: [...]

I just tried, but it still doesn't work(also, I was having 1920 set as width before, and it was still taking 640).

jgrossrieder, 2021-01-10:
I think I've a lead: The "cameraPreview" class that I provide in the camera preview options is applied, but it does not take in account the css isolation name(like apparently, another class that uses this "cameraPreview" class ends up by having a class named "cameraPreview[_ngcontent-ofb-c184]". So it doesn't get the width/height set properly.

If I put my CSS class in global.scss, which isn't renamed due to being... global. it now works.

Any idea how to make it work within my page scss ?

pbowyer, 2021-12-16:
I'm closing as this is now nearly a year old. Hope you worked it out!

closed 2021-12-16 by pbowyer