capacitor-community/camera-preview #92: Capacitor 3 beta support ?

togro, 2021-02-07:
**Describe the bug**
camera-preview has support for capacitor 3 beta ?

On Android, I have modified MainActivity.java for this :

```java
package vigiapp.movilgate.com;

import android.os.Bundle;

import com.ahm.capacitor.camera.preview.CameraPreview;
import com.getcapacitor.BridgeActivity;

public class MainActivity extends BridgeActivity {
  @Override
  public void onCreate(Bundle savedInstanceState) {
    super.onCreate(savedInstanceState);
       registerPlugin(CameraPreview.class);
  }
}

```

But AndroidStudio said that method doesn't exists: **registerPlugin**

May be capacitor 3 is not supported yet ?

arielhernandezmusa, 2021-02-08:
Not supported yet, on it ...

arielhernandezmusa, 2021-02-08:
this version aims to add support for ios for capacitor 3 @capacitor-community/camera-preview-2.0.0-beta.0 (https://github.com/capacitor-community/camera-preview/tree/capacitor-3-support)

Linknpay, 2021-05-21:
Hi, capacitor has launched their final release 3.0.0
Could you update the plugin because we are obtaining this error : 
```
Error: node_modules/@capacitor/core/types/definitions-internal.d.ts:16:18 - error TS2430: Interface 'CapacitorInstance' incorrectly extends interface 'CapacitorGlobal'.
  Types of property 'Plugins' are incompatible.
    Type '{ [pluginName: string]: { [prop: string]: any; }; }' is missing the following properties from type 'PluginRegistry': CameraPreview
```

pbowyer, 2021-12-16:
Hi! We will very shortly have a Capacitor 3.0 compatible release. Please keep an eye on #183 - I'm closing this issue in favour of it.

closed 2021-12-16 by pbowyer