capacitor-community/camera-preview #97: Follow SemVer

ryaa, 2021-02-16:
**Is your feature request related to a problem? Please describe.**
Regular "minor versions" (e.g 1.0.3+ where toBack option was added with the default to false, while it was true before) break compatibility with existing codebases.

**Describe the solution you'd like**
Follow SemVer and change major version number if there is a breaking change.

**Describe alternatives you've considered**
Document each release changes and specifically note about breaking changes.

**Additional context**
N/A

arielhernandezmusa, 2021-02-28:
Agree, thank you @ryaa

jcesarmobile, 2023-06-08:
The plugin seems to be following semver now with 4.x version supporting Capacitor 4 and the next major version will be 5.x to also match Capacitor 5

closed 2023-06-08 by jcesarmobile