capacitor-community/camera-preview #103: Adding import to Android

BLTowsen, 2021-02-26:
When adding the required import to Android it is not finding the package.

I am probably doing it incorrectly but when adding:

```
import com.ahm.capacitor.camera.preview.CameraPreview;

this.init(savedInstanceState, new ArrayList<Class<? extends Plugin>>() {{
      // Additional plugins you've installed go here
      // Ex: add(TotallyAwesomePlugin.class);
      add(CameraPreview.class);
}});
```
It gives an error for the package name.  Is there any extra steps I need to follow before importing the package.

BLTowsen, 2021-03-07:
I was dumb and didn't realize the package you need to import is in the android folder of the app.

closed 2021-03-07 by BLTowsen

zNoobs, 2021-03-22:
Can you explain?
Because the package does not exist...
`import com.ahm.capacitor.camera.preview.CameraPreview;
`