capacitor-community/camera-preview #110: Android - Orientation of B64 image upside down with Front Facing Camera

mattmilan-dev, 2021-03-14:
**Describe the bug**
When taking a photograph on the Samsung Galaxy A40 using the front facing camera on device, the finalised base64 image appears upside down, even after the supposed fix of #75.

**To Reproduce**
Load the camera preview and run `await CameraPreview.capture();` displaying the image in an img tag.

**Expected behavior**
The expected behaviour is that regardless of camera, the orientation should always be the right way up.

**Screenshots**
[image]
[image]

**Smartphone (please complete the following information):**
 - Device: Samsung Galaxy A40
 - OS: Android 10

**Additional context**
Oddly, the device has 3 cameras available (even though it only has a front and back facing camera in the official camera app): Back, Front (Really Zoomed In) and Front (normal). To get to the front facing camera i have to call the flip function twice, which led me to believe that the `cameraCurrentlyLocked == Camera.CameraInfo.CAMERA_FACING_FRONT` part of the `if` statement on L658 was blocking reaching the `switch` statement. Once i removed that part of the `if` statement it worked as expected.

I do not know Java so i don't know what is at fault here.

facusatoshi, 2021-08-17:
Im getting the same error on Samsung s10e

TreehouseNorris, 2023-05-17:
this fixed it for me on android 
```
CameraPreview.start({
        disableExifHeaderStripping: false // true by default
})
```