capacitor-community/camera-preview #111: "toBack" does not work on iOS

amylius, 2021-03-15:
**Describe the bug**
Simply put, when toBack is set to false, the camera preview displays (on IOS and WEB) but when using toBack, it still succeeds on web but fails on IOS. Yes, everything has 'background: transparent'. I don't have an Android phone handy to know whether or not that works, but I can confirm via logs that the camera is running (and on the iphone has the green dot indicating that it is running). It just remains completely black in the background. 

The code:

my HTML file:
```
<ion-content  id="content"  class="transparent" style="background: transparent !important; background-color: transparent !important;">
  <div class="red_button">
    <ion-icon (click)="endVideo()" class="end_icon" name="close-circle-outline"></ion-icon>
  </div>
</ion-content>
```

My ts file:
```
  public cameraPreviewOpts: CameraPreviewOptions = {
    position: this.cameraDirection,
    parent: 'content',
    toBack: true
  };

{...}

CameraPreview.start(this.cameraPreviewOpts).then(
      (res) => {
        console.log(res)
        console.log("WORKING")
      },
      (err) => {
        console.log(err)
        console.log("FAILED")
      });
```

Not sure what else I can do but I've tried everything I can think of. Let me know if anyone has any suggestions. Thanks.

tafelnl, 2021-03-17:
What you could do to discover what divs are responsible for it, is the following:

Inspect your webapp in your browser. Add some very noticeable background color to <html>. Then delete the most upper root div and see if you can see the background color of <html>. You do? Nice. Undo the deletion and remove the child of that div. Etc etc. This way you can gradually debug and discover what divs are responsible for overlaying the barcode scanner.

pbowyer, 2021-12-16:
@tafelnl is correct. I'm closing this ticket; best of luck with your app!

closed 2021-12-16 by pbowyer