capacitor-community/camera-preview #113: Black Screen on IoS - PWA

Christophe-re, 2021-03-18:
**Describe the bug**
The camera not working on Safari and Chrome on IOS 14, I have a black screen.
 This error is in the console :
Unhandled Promise Rejection: NotAllowedError: The request is not allowed by the user agent or the platform in the current context, possibly because the user denied permission.

All is good on Mac os.

**To Reproduce**

Html : 
`<div class="container">
</div>
<div id="cameraPreview"></div>
<button  (click)="startCamera()">test</button>`

ts : 
`
@Component({
  selector: 'app-capacitor',
  templateUrl: './capacitor.component.html',
  styleUrls: ['./capacitor.component.scss']
})
export class CapacitorComponent implements OnInit {
  @ViewChild('cameraPreview', {static: true}) element: ElementRef;

  constructor() { 
  }

  ngOnInit() {}
  
  public startCamera() {
    CameraPreview.start({
      position: 'rear',
      parent: 'cameraPreview', 
      className: 'cameraPreview',
      disableAudio: true,
      width: 300,
      height: 600,
    });
  }
}`

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone 8]
 - OS: [IOS 14]
 - Browser [safari, chrome]

argometric, 2021-06-01:
add autoplay and playsinline to the video tag after its generated, this should fix the black screen issue. 

CameraPreview.start({
        position: "rear",
        toBack: true,
        parent: "content",
        heigth: 300,
        width: 400,
        disableAudio: true
      }).then(() => {
        document.getElementById("video")?.setAttribute("playsinline", "true");
        document.getElementById("video")?.setAttribute("autoplay", "true");
      })

that should fix it. I got issue now that the screen on iOs gets stuck every second for a short period of time, like its lagging. If you find a solution for that please let me know.

pbowyer, 2021-12-16:
Closing this ticket as there's been no recent feedback. Please submit a pull request if code changes are needed, or re-open this ticket.

closed 2021-12-16 by pbowyer