capacitor-community/camera-preview #114: Cannot resolve com.ahm.capacitor.camera.preview.CameraPreview; on android

zNoobs, 2021-03-22:
I followed exactly the readme, but this code won't work:
```
import com.ahm.capacitor.camera.preview.CameraPreview;
this.init(savedInstanceState, new ArrayList<Class<? extends Plugin>>() {{
      // Additional plugins you've installed go here
      // Ex: add(TotallyAwesomePlugin.class);
      add(CameraPreview.class);
}});
```
Because Android Studio doesn't find the package _com.ahm.capacitor.camera.preview.CameraPreview_
What is missing?

philippendrulat, 2021-03-23:
I had the same problem initially, I fixed it with a complete reload of the project (removing .idea folder in the android folder) and reopening the project with android studio. 
This issue might also be a problem when the plugin was not correctly installed with `npm install` following `npx cap sync android`

zNoobs, 2021-03-23:
I did:
- delete `myProject/android/.idea` folder
- `npm install @capacitor-community/camera-preview`
- `ionic build`
- `npx cap sync android` (the 2 previous commands can be replaced by `ionic capacitor sync android`)
- `npx cap open android`
And now it works!
But I did (almost) same before (`npx cap sync` instead of `npx cap sync android` because the doc says `npx cap sync`) so I don't get why it didn't work at first. I should have thought first to "reset" the android project dependencies, thanks!

PS: I had a small problem, my AndroidManifest.xml was saying I don't have my MainActivity anymore (which I still had) so I had to click on the error and "create" it from the error popup, it didn't create it but the error was gone then I could compile.

closed 2021-03-23 by zNoobs