capacitor-community/camera-preview #115: The camera preview is over all the HTML elements on Android

zNoobs, 2021-03-23:
Hi,

**Version**
```
"@angular/core": "~11.2.0",
"@ionic/angular": "^5.5.2",
"@ionic-native/core": "^5.31.1",
"@capacitor-community/barcode-scanner": "^1.2.1",
"@capacitor/android": "^2.4.6",
 "@capacitor/core": "2.4.6",
```
**Smartphone (please complete the following information):**
 - Device: Xiaomi Mi A1
 - OS: Android 9

I'm using Ionic 5 with Angular and Capacitor.
On my Android phone the camera preview is over all the HTML elements and impossible to change the opacity or z-index, it doesn't do anything, if I use:
```
import { Plugins } from "@capacitor/core"
const { CameraPreview } = Plugins;
```
But if I use this, it works but the camera takes few seconds to open:
`import { CameraPreview } from '@capacitor-community/camera-preview';`

The doc says the first one, but it doesn't work correctly with this one (camera preview is over everything).
On the web, both works well.

### **Which one should I be using?**

mic8, 2021-03-24:
I have same issue for ionic 5, and after using this:

`import { CameraPreview } from '@capacitor-community/camera-preview';`

It works but takes few seconds to open, and cannot use any native features like flip or flash.

I also try toBack option for my cases, but all HTML elements is over the camera preview so there is screen shows for the camera preview.

Thanks

zNoobs, 2021-03-24:
I finally re-installed the module "@capacitor-community/camera-preview" and now it's version 1.2.1.

`import { CameraPreview } from '@capacitor-community/camera-preview';`
uses the web implementation, which doesn't support flash and maybe not flip...

Now using the Plugin implementation (android) with the option "toBack" it works for me!

Did you set the background to transparent? 
Here some simplified parts of my code, it might help you:
- my scss file
```
ion-content {
   --background: transparent !important;
}
.camera-preview {
  position: absolute;
  z-index: 1;
}
.overlay-hole {
  position: absolute;
  width: 500;
  height: 500;
  top: 50%;
  left: 50%;
  transform: translate(-50%, -50%);
  border: 1px solid #BFBFBF;
  box-shadow: 0 0 0 9999px rgba(0, 0, 0, 0.2);
  z-index: 999999998;
}
```
- my html:
```
<ion-content>
  <div>
    <div id="divCameraPreviewContent">
      <div id="divOverlayHole" class="overlay-hole"></div>
    </div>
  </div>
</ion-content>
```
- my ts:
```
//the imports
import { Plugins } from "@capacitor/core"
const { CameraPreview } = Plugins;
import { CameraPreviewOptions, CameraPreviewPictureOptions, CameraPreviewFlashMode } from '@capacitor-community/camera-preview';
...
//I call this after ionViewDidEnter()
  const cameraPreviewOptions: CameraPreviewOptions = {
        position: 'rear',
        height: window.screen.height,
        parent: 'divCameraPreviewContent',
        className: 'camera-preview',
        disableAudio: true,
        toBack: true
      };
      CameraPreview.start(cameraPreviewOptions);
```
To manage the flash to work, I don't follow the documentation, I use:
`  CameraPreview.setFlashMode({ flashMode: 'on' });`

Let me know if it works for you then I'll close the issue.
I have a new problem, when I take picture from front camera (selfie camera), the picture is rotated by 90°, I don't know if it's device specific or if it's the normal behavior of this plugin...

BLTowsen, 2021-03-24:
I have tried the above and in the logs it shows a Camera Preview is starting but is still hidden behind the background.  Are you starting the camera inside ionViewDidEnter()?  I ensured my version is the same and can send through more info if needed.

mic8, 2021-03-24:
> I finally re-installed the module "@capacitor-community/camera-preview" and now it's version 1.2.1.
>
> `import { CameraPreview } from '@capacitor-community/camera-preview';`
> uses the web [...]

I already try this one, but there is another issue after i run it on android:

here is the logcat:

```
2021-03-24 23:05:28.797 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:28.810 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:28.818 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:28.831 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:28.839 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:28.848 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:28.859 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:28.869 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:28.880 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:28.896 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:28.905 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:28.951 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:28.956 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:28.964 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:28.972 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:28.981 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:28.986 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:28.994 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.001 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.009 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.015 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.022 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.082 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.092 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.101 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.109 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.118 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.127 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.134 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.145 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.154 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.164 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.173 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.183 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.193 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.201 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.362 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.367 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.374 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.379 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.382 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.387 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.391 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.396 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.400 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.404 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.409 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.413 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.417 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.422 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.431 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.440 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.448 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.461 21507-21626/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:29.468 21507-21626/id.ammana.app E/CameraDevice-JV-1: OPPO_NIGHT_CEHCK packageName:id.ammana.app str:com.oppo.camera
2021-03-24 23:05:29.508 21507-21748/id.ammana.app E/cr_VideoCapture: CameraDevice.StateCallback onOpened
2021-03-24 23:05:29.511 21507-21748/id.ammana.app E/libc: Access denied finding property "vendor.debug.bq.dump"
2021-03-24 23:05:29.512 21507-21748/id.ammana.app E/libc: Access denied finding property "vendor.debug.bq.dump"
2021-03-24 23:05:29.512 21507-21748/id.ammana.app E/libc: Access denied finding property "vendor.debug.bq.dump"
2021-03-24 23:05:29.523 21507-21748/id.ammana.app E/CheckPermission: camera-code= 5
[... cut ...]
2021-03-24 23:05:30.600 21507-21836/id.ammana.app E/libc: Access denied finding property "vendor.debug.bq.dump"
2021-03-24 23:05:30.601 21507-21836/id.ammana.app E/libc: Access denied finding property "vendor.debug.bq.dump"
2021-03-24 23:05:30.601 21507-21836/id.ammana.app E/libc: Access denied finding property "vendor.debug.bq.dump"
2021-03-24 23:05:30.606 21507-21836/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:30.611 21507-21836/id.ammana.app E/CheckPermission: camera-code= 5
2021-03-24 23:05:31.036 21507-21836/id.ammana.app E/CheckPermission: camera-code= 5

```

zNoobs, 2021-03-24:
@BLTowsen I call it in ionViewDidEnter(), try with the option `toBack: false`, are you sure you set your background to `transparent !important` ? Maybe you have some background on your ion-app ?
Try to create a new page and just add the code I wrote in my previous message

@mic8 You have to add the permissions to your AndroidManifest.xml:
```
    <!-- Camera, Photos, input file -->
    <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE"/>
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
    <!-- Video -->
    <uses-permission android:name="android.permission.CAMERA" />
    <!-- Audio -->
    <uses-permission android:name="android.permission.RECORD_AUDIO" />
    <uses-permission android:name="android.permission.MODIFY_AUDIO_SETTINGS"/>
    <uses-feature android:name="android.hardware.camera" />
```
and also check the Camera permission on your MainActivity.java like this :
```
import com.ahm.capacitor.camera.preview.CameraPreview;

public class MainActivity extends BridgeActivity {
  @Override
  public void onCreate(Bundle savedInstanceState) {
    super.onCreate(savedInstanceState);

    if (ContextCompat.checkSelfPermission(this, Manifest.permission.CAMERA) != PackageManager.PERMISSION_GRANTED) {
      ActivityCompat.requestPermissions(this, new String[] {Manifest.permission.CAMERA},50);
    }

    this.init(savedInstanceState, new ArrayList<Class<? extends Plugin>>() {{
      add(CameraPreview.class);
    }});
  }
}
```
Check also the permissions granted on your app for your app (go on the App info / Permissions), maybe you disabled Camera permission, you might also need audio and write files.

BLTowsen, 2021-03-25:
This is the logs I am getting from logcat.  It shows the camera is starting and should be visible.

```
2021-03-25 13:06:17.695 13230-14031/io.fieldtech.mediaLibrary V/Capacitor/Plugin: To native (Capacitor plugin): callbackId: 55526378, pluginId: CameraPreview, methodName: start
2021-03-25 13:06:17.696 13230-14031/io.fieldtech.mediaLibrary V/Capacitor: callback: 55526378, pluginId: CameraPreview, methodName: start, methodData: {"position":"rear","height":800,"parent":"divCameraPreviewContent","className":"camera-preview","disableAudio":true,"toBack":true}
2021-03-25 13:06:17.714 13230-13989/io.fieldtech.mediaLibrary D/Capacitor: Handling local request: http://localhost/assets/icon/favicon.png
2021-03-25 13:06:17.755 13230-13230/io.fieldtech.mediaLibrary E/libc: Access denied finding property "vendor.camera.aux.packagelist"
2021-03-25 13:06:17.752 13230-13230/io.fieldtech.mediaLibrary W/ch.mediaLibrary: type=1400 audit(0.0:10043): avc: denied { read } for name="u:object_r:vendor_default_prop:s0" dev="tmpfs" ino=925 scontext=u:r:untrusted_app:s0:c194,c256,c512,c768 tcontext=u:object_r:vendor_default_prop:s0 tclass=file permissive=0
2021-03-25 13:06:17.768 13230-13230/io.fieldtech.mediaLibrary E/libc: Access denied finding property "vendor.camera.aux.packagelist"
2021-03-25 13:06:17.765 13230-13230/io.fieldtech.mediaLibrary W/ch.mediaLibrary: type=1400 audit(0.0:10044): avc: denied { read } for name="u:object_r:vendor_default_prop:s0" dev="tmpfs" ino=925 scontext=u:r:untrusted_app:s0:c194,c256,c512,c768 tcontext=u:object_r:vendor_default_prop:s0 tclass=file permissive=0
2021-03-25 13:06:17.784 13230-13230/io.fieldtech.mediaLibrary D/AppResolutionTuner: loadTunerAppList + 
2021-03-25 13:06:17.790 13230-13230/io.fieldtech.mediaLibrary D/AppResolutionTuner: loadTunerAppList - 
2021-03-25 13:06:17.791 13230-13230/io.fieldtech.mediaLibrary D/SurfaceExt: initResolutionTunner, mPackageName:io.fieldtech.mediaLibrary,mContainPackageName:falsemXScaleValue:1.0,mYScaleValue:1.0
2021-03-25 13:06:17.807 13230-13230/io.fieldtech.mediaLibrary E/libc: Access denied finding property "vendor.camera.aux.packagelist"
2021-03-25 13:06:17.805 13230-13230/io.fieldtech.mediaLibrary W/ch.mediaLibrary: type=1400 audit(0.0:10045): avc: denied { read } for name="u:object_r:vendor_default_prop:s0" dev="tmpfs" ino=925 scontext=u:r:untrusted_app:s0:c194,c256,c512,c768 tcontext=u:object_r:vendor_default_prop:s0 tclass=file permissive=0
2021-03-25 13:06:17.808 13230-13230/io.fieldtech.mediaLibrary V/CameraBase: connect: connect
2021-03-25 13:06:17.921 13230-13230/io.fieldtech.mediaLibrary D/Camera-JNI: (tid:13230)[MtkJNICameraContext] this:0x7a99d78ec0 camera->getStrongCount(3) 
2021-03-25 13:06:17.927 13230-13230/io.fieldtech.mediaLibrary V/Camera-JNI: getParameters
2021-03-25 13:06:17.927 13230-13230/io.fieldtech.mediaLibrary V/Camera-JNI: get_native_camera: context=0x7a99d78ec0, camera=0x7a9a320180
2021-03-25 13:06:17.927 13230-13230/io.fieldtech.mediaLibrary V/Camera: getParameters
2021-03-25 13:06:17.931 13230-13230/io.fieldtech.mediaLibrary E/libc: Access denied finding property "vendor.camera.aux.packagelist"
2021-03-25 13:06:17.935 13230-13230/io.fieldtech.mediaLibrary D/Preview: screen is rotated 0deg from natural
2021-03-25 13:06:17.935 13230-13230/io.fieldtech.mediaLibrary D/Preview: back camera is oriented -90deg from natural
2021-03-25 13:06:17.935 13230-13230/io.fieldtech.mediaLibrary D/Preview: need to rotate preview 90deg
2021-03-25 13:06:17.935 13230-13230/io.fieldtech.mediaLibrary V/Camera-JNI: setDisplayOrientation
2021-03-25 13:06:17.935 13230-13230/io.fieldtech.mediaLibrary V/Camera-JNI: get_native_camera: context=0x7a99d78ec0, camera=0x7a9a320180
2021-03-25 13:06:17.935 13230-13230/io.fieldtech.mediaLibrary V/Camera: sendCommand
2021-03-25 13:06:17.936 13230-13230/io.fieldtech.mediaLibrary V/Camera-JNI: getParameters
2021-03-25 13:06:17.936 13230-13230/io.fieldtech.mediaLibrary V/Camera-JNI: get_native_camera: context=0x7a99d78ec0, camera=0x7a9a320180
2021-03-25 13:06:17.936 13230-13230/io.fieldtech.mediaLibrary V/Camera: getParameters
2021-03-25 13:06:17.937 13230-13230/io.fieldtech.mediaLibrary V/Camera-JNI: getParameters
2021-03-25 13:06:17.937 13230-13230/io.fieldtech.mediaLibrary V/Camera-JNI: get_native_camera: context=0x7a99d78ec0, camera=0x7a9a320180
2021-03-25 13:06:17.937 13230-13230/io.fieldtech.mediaLibrary V/Camera: getParameters
2021-03-25 13:06:17.940 13230-13230/io.fieldtech.mediaLibrary V/Camera-JNI: setParameters
2021-03-25 13:06:17.940 13230-13230/io.fieldtech.mediaLibrary V/Camera-JNI: get_native_camera: context=0x7a99d78ec0, camera=0x7a9a320180
2021-03-25 13:06:17.940 13230-13230/io.fieldtech.mediaLibrary V/Camera: setParameters
2021-03-25 13:06:17.942 13230-13230/io.fieldtech.mediaLibrary I/System.out: camera started
2021-03-25 13:06:17.943 13230-13230/io.fieldtech.mediaLibrary D/CameraActivity: cameraCurrentlyLocked:0
2021-03-25 13:06:17.952 13230-13230/io.fieldtech.mediaLibrary D/Preview: optimal preview size: w: 1920 h: 1440
2021-03-25 13:06:17.953 13230-13230/io.fieldtech.mediaLibrary D/Preview: optimal preview size: w: 1920 h: 1440
2021-03-25 13:06:17.965 13230-13230/io.fieldtech.mediaLibrary D/Preview: center horizontally
2021-03-25 13:06:17.966 13230-13230/io.fieldtech.mediaLibrary D/layout: left:-240
2021-03-25 13:06:17.966 13230-13230/io.fieldtech.mediaLibrary D/layout: top:0
2021-03-25 13:06:17.966 13230-13230/io.fieldtech.mediaLibrary D/layout: right:960
[... cut ...]
  top: 50%;
  left: 50%;
  transform: translate(-50%, -50%);
  border: 1px solid #bfbfbf;
  box-shadow: 0 0 0 9999px rgba(0, 0, 0, 0.2);
  z-index: 999999998;
}

```
**sentry.page.html**
```
<ion-content>
  <div>
    <div id="divCameraPreviewContent">
      <div id="divOverlayHole" class="overlay-hole"></div>
    </div>
  </div>
</ion-content>
```

When running the application all I see is a light grey screen and no camera.

If I set `toBack=false` then the camera image covers the whole screen.

zNoobs, 2021-03-25:
I don't know if you can set a background on the ion-app but maybe try in your global.scss:
```
ion-app{
  --background: transparent !important;
}
```
Also try to set all the backgrounds to transparent on the body and html in your sentry.page.scss:
```
html, body {
--ion-content-background-color: transparent !important;
--ion-background-color: transparent !important;
--background: transparent !important;
background: transparent !important;
}
```

BLTowsen, 2021-03-25:
I finally got it working!!!!

After some testing I found the issue is with the `:host`  background

The fix was using
```
ion-content {
  --background: transparent !important;
}
```
in the `global.scss` file because as far as I can see :host can not be set in ionic