capacitor-community/camera-preview #124: Removed WRITE_EXTERNAL_STORAGE and READ_EXTERNAL_STORAGE permissions (Android)

ryaa, 2021-04-19:
**Is your feature request related to a problem? Please describe.**

My assumption is that this plugin does not need to read or write to the external storage therefore WRITE_EXTERNAL_STORAGE and READ_EXTERNAL_STORAGE permissions are not required - see
https://developer.android.com/reference/android/Manifest.permission#WRITE_EXTERNAL_STORAGE and https://developer.android.com/reference/android/Manifest.permission#READ_EXTERNAL_STORAGE

**Describe the solution you'd like**
Removed WRITE_EXTERNAL_STORAGE and READ_EXTERNAL_STORAGE permissions from the manifest file

**Describe alternatives you've considered**
N/A

**Additional context**
The Google Play informs that starting May 5th, you must let us know why your app requires broad storage access
```
We've detected that your app contains the requestLegacyExternalStorage flag in the manifest file of 1 or more of your app bundles or APKs.

Developers with apps on devices running Android 11+ must use Scoped Storage to give users better access control over their device storage. To release your app on Android 11 or newer after May 5th, you must either:

Update your app to use more privacy friendly best practices, such as the Storage Access Framework or Media Store API
Update your app to declare the All files access (MANAGE_EXTERNAL_STORAGE) permission in the manifest file, and complete the All files access permission declaration in Play Console from May 5th
Remove the All files access permission from your app entirely
For apps targeting Android 11, the requestLegacyExternalStorage flag will be ignored. You must use the All files access permission to retain broad access.

Apps requesting access to the All files access permission without a permitted use will be removed from Google Play, and you won't be able to publish updates.
```

requestLegacyExternalStorage is often used with WRITE_EXTERNAL_STORAGE so it might require removing the both (if they are not required)

Please also see https://support.google.com/googleplay/android-developer/answer/10467955

ryaa, 2021-04-19:
I added the appropriate PR - see https://github.com/capacitor-community/camera-preview/pull/123

pbowyer, 2021-12-16:
Closing as this was merged. Thanks for your contributions 😄

closed 2021-12-16 by pbowyer