capacitor-community/camera-preview #129: Don't have auto rotation

ManjorStreep, 2021-04-28:
**Describe the bug**
If you lock a device rotation and capture one image in different rotation, the image won't rotate. This is shown with wrong orientation.

**To Reproduce**
Steps to reproduce the behavior:
1. Lock rotation android.
2. Open the camera
3. Capture image
4. Move the device with another position o rotation
5. See the image with wrong orientation.

**Screenshots**
Image with 90 degrees of rotation in device 
[image]
Expected image
[image]

**Smartphone (please complete the following information):**
 - Device: Android +5

riderx, 2022-03-02:
got the same issue.
if you go to Preview.java
and comment `setCameraDisplayOrientation();` it fix the issue on my side

luizpbello, 2026-09-30:
do you have any solution?