capacitor-community/camera-preview #136: Video on iOs

Linknpay, 2021-05-14:
When will it possible to use record video on iOs ?
Thank you

rudyZanotti, 2021-06-01:
I would love this feature !

rudyZanotti, 2021-06-03:
Hello @Linknpay , i just begin this feature. If you want to improve it you can fork it 👍 

https://github.com/rudyZanotti/camera-preview/tree/start-recording-ios

Kaoschuks, 2021-06-06:
@rudyZanotti  nice work man.
Been waiting for this for a while now.

You can look at https://github.com/cordova-plugin-camera-preview/cordova-plugin-camera-preview to use its iOS code base to make it better. I really wish I can code in iOS would have definitely helped

Codemax999, 2021-08-16:
@rudyZanotti any update on this? Would be awesome to get this on iOS

Codemax999, 2021-09-13:
@Kaoschuks were you able to get the video recording working for iOS?

Kaoschuks, 2021-09-14:
using this plugin no @Codemax999

Codemax999, 2021-09-14:
> using this plugin no @Codemax999

Were you able to do ios video with html overlay on any other plugins? I've only seen the default video recorder. Thanks

Kaoschuks, 2021-09-15:
I am experimenting with something on 

https://github.com/Kaoschuks/ionic-media-editor

@Codemax999 you can check it out

mesqueeb, 2021-12-03:
It would be really cool if we could have the functions `startRecordVideo` and `stopRecordVideo` to work on iOS:
[image]

From the readme:
https://github.com/capacitor-community/camera-preview#startrecordvideooptions-------android-only

@pbowyer do you have any skills in Swift to make this possible? Or do you know anyone who can? We can sponsor this feature.

Is it possible to open a fund for this feature specifically? Eg. on issue hunt or something?

pbowyer, 2021-12-03:
It's great to hear you can sponsor this! I don't have the skills so I'm paging @nick-krantz - is this something you'd like to take on?

nick-krantz, 2021-12-03:
@pbowyer, @mesqueeb Thanks for thinking of me! I unfortunately don't feel confident in my swift abilities to flush out video recording for iOS to take this work up.

RobinGiel, 2022-04-13:
I see that in the past this was possible with the Last Cam plugin -> https://github.com/Bengejd/cordova-plugin-last-cam/blob/master/src/ios/LastCam.swift

@nick-krantz would this help?

RobinGiel, 2022-10-11:
For anyone who is still interested in making video rec work on iOs. Here is a react native plugin that achieved that. Maybe we can use the same code for this plugin on the native side? https://github.com/mrousavy/react-native-vision-camera