capacitor-community/camera-preview #139: Capacitor 3 import Error  Interface 'CapacitorInstance' incorrectly extends interface 'CapacitorGlobal'.Types of property 'Plugins'

Scan0815, 2021-05-29:
**Describe the bug**
Not working on Capacitor 3, on Build i get this error in console. 

[ ERROR ]  TypeScript: ./node_modules/@capacitor/core/types/definitions-internal.d.ts:16:18
           Interface 'CapacitorInstance' incorrectly extends interface 'CapacitorGlobal'.Types of property 'Plugins'
           are incompatible.Property 'CameraPreview' is missing in type '{ [pluginName: string]: { [prop: string]: any;
           }; }' but required in type 'PluginRegistry'.

     L15:   */
     L16:  export interface CapacitorInstance extends CapacitorGlobal {
     L17:      /**

Hope that´s help. Sheers Oliver

Scan0815, 2021-05-29:
I test the capacitor 3 support branch too, 
`[ ERROR ]  TypeScript: ./node_modules/@capacitor/core/types/definitions-internal.d.ts:16:18
           Interface 'CapacitorInstance' incorrectly extends interface 'CapacitorGlobal'.Types of property 'Plugins'
           are incompatible.Property 'CameraPreview' is missing in type '{ [pluginName: string]: { [prop: string]: any;
           }; }' but required in type 'PluginRegistry'.

     L15:   */
     L16:  export interface CapacitorInstance extends CapacitorGlobal {
     L17:      /**
`

Kaoschuks, 2021-06-06:
please can this be fixed am currently using nix/webcam to replace this plugin till the support for capacitor 3 is working.
Please I really love this feature

Kaoschuks, 2021-06-06:
Please can this be fixed quickly. I have an app that uses this and currently can continue to compile after upgrading to capacitor 3.

I had to switch to ngx-webcam for the time been . Please lets get this done

Bforsyth1234, 2021-06-06:
plus one this

RobertoGemo, 2021-06-07:
Hi,
I have capacitor 2.4.7, but in the package.json if this project there are peer dependencis to "latest" capacitor so it will always throw this error

nicolasrhone, 2021-06-07:
Same problem here, the plugin is not updated for Capacitor 3.
I discovered it after updating all my project and am now stuck :-)

anth0, 2021-06-11:
A capacitor 3 update is on its way : https://github.com/anth0/camera-preview/tree/capacitor-3-upgrade. 
I just need to figure out what's going on with one of the latest merge about video recording because it does not compile on my hand.

gamanuel, 2021-06-13:
Has anyone solve this issue?

Mathyn, 2021-06-14:
I've added a PR with a fix https://github.com/capacitor-community/camera-preview/pull/146

sinkingtitanic, 2021-06-15:
@Mathyn I tested your PR and it works. @bialabs please merge 

For anyone who wants a quick fix until authors merge, just install @Mathyn [PR](https://github.com/Mathyn/camera-preview): 
```
npm uninstall @capacitor-community/camera-preview
npm install git+https://github.com/Mathyn/camera-preview.git
```

That's all folks!

Kaoschuks, 2021-07-07:
I have created a repo here that uses this plugin and other methods for getting camera data

meforce, 2021-08-05:
I recommend you to take a look 
https://github.com/capacitor-community/camera-preview/issues/151#issuecomment-893511569

ryaa, 2021-09-13:
> https://github.com/Mathyn/camera-preview.git

The plugin builds when installed from this repo however it asks for microphone permission still, which is not needed.

pbowyer, 2021-12-16:
There will shortly be a Capacitor 3 build on NPM - please keep an eye on #183 for progress (I don't have permission to upload it). I'm closing this issue now, but thank you all for your contributions! Let's see how good we can make camera-preview in 2022.

closed 2021-12-16 by pbowyer