capacitor-community/camera-preview #140: capacitor version depencies

bialabs, 2021-06-01:
**Describe the bug**
In my project i'm using capacitor 2.4.7, but npm downloads version 3.0 dependecy because in the package.json of this package is specified "latest". This causes typescript errors like "Interface 'CapacitorInstance' incorrectly extends interface 'CapacitorGlobal'"
For now i've fixed by setting manually 2.4.7 instead of 3.0 in package-lock.json under camerapreview -> dependencies
 
**To Reproduce**
Install camera preview in a capacitor 2 project

anth0, 2021-06-11:
@bialabs You are right. Please submit a PR.

Mathyn, 2021-06-14:
I've added a PR https://github.com/capacitor-community/camera-preview/pull/146

bialabs, 2021-06-14:
@Mathyn in [this commit](https://github.com/martinkasa/capacitor-secure-storage-plugin/commit/b8eb1d8aa0c1b06a1af778cadbc2feb70f6ebe19#diff-7ae45ad102eab3b6d7e7896acd08c427a9b25b346470d7bc6507b6481575d519) i see fixes also in the /ios folder. Do you think are needed also for this plugin?

Mathyn, 2021-06-14:
Good one. I did a quick check:

`ios/Podfile.lock` -> shows version 2.0.1 currently, this one has not been updated for some time. It might be fine but someone might need to test this (don't have a Macbook at hand).

Bunch of files in `ios/Pods` were changed but this directory does not exist for this project by default

bialabs, 2021-06-14:
Yeah, for this reason i didn't added a PR
@arielhernandezmusa can you check?

sinkingtitanic, 2021-06-15:
@Mathyn I tested your PR and it works. @bialabs please merge 

For anyone who wants a quick fix until authors merge, just install @Mathyn [PR](https://github.com/Mathyn/camera-preview): 
```
npm uninstall @capacitor-community/camera-preview
npm install git+https://github.com/Mathyn/camera-preview.git
```

That's all folks!

Kaoschuks, 2021-07-07:
I have created a repo here that uses this plugin and other methods for getting camera data

https://github.com/Kaoschuks/ionic-media-editor

Please review

pbowyer, 2022-03-06:
Closing as this was fixed.

closed 2022-03-06 by pbowyer