capacitor-community/camera-preview #142: captureSample is not a function

fr3shn3ss, 2021-06-06:
**Describe the bug**
Uninstalled and installed the latest version (1.2.1) using npm.  

"captureSample" function does not exist in "node_modules\@capacitor-community\camera-preview\android\src\main\java\com\ahm\capacitor\camera\preview\CameraPreview.java" 

Capacitor version: 
    "@capacitor/android": "^2.1.2",
    "@capacitor/core": "2.1.2",

ufocruz, 2021-07-20:
Same problem

ufocruz, 2021-07-28:
at the moment, my work arround is installing with this command:
`npm install git+https://github.com/capacitor-community/camera-preview.git`

hodeware, 2021-07-31:
> at the moment, my work arround is installing with this command:
> `npm install git+https://github.com/capacitor-community/camera-preview.git`

That saved me days. Hi Guys, is it possible to publish that to the NPM, please?

jcesarmobile, 2023-06-08:
Can't reproduce on version 4 of the plugin

closed 2023-06-08 by jcesarmobile