capacitor-community/camera-preview #143: Zoom

glancel, 2021-06-06:
Hello everyone!

I love this plugin and I would like to thank anyone who contributed to the project.
Has anyone experimented with something I could work on for the zoom feature?

Thank you

mehdis34, 2021-07-09:
Hi,
 I have the same problem as you.
 Have you found a solution?

glancel, 2021-07-09:
Actually no, but it's an important feature but I can't get it to work... If you find any alternative solution please let me know.

Alfer-Star, 2021-09-02:
There is a Cordova Plugin which is like this one and has zoom.
See also #99 
Last time (in march 2021) I tested, the to-back feature does not work correctly with capacitor.

lauritaila, 2021-11-30:
is the height and width if you erase that in yhe CameraPreviewOptions the camera should look fine.

pbowyer, 2022-03-06:
Hi everyone, pinch and zoom is now available in version 2.1.0 released today. Good luck using it!

closed 2022-03-06 by pbowyer