capacitor-community/camera-preview #148: Camera preview not working on Android

sinkingtitanic, 2021-06-15:
**Describe the bug**
Camera preview not working on Android. Please see screenshot blow 

**To Reproduce**
MainActivity File: 
```package io.ionic.starter;
import com.ahm.capacitor.camera.preview.CameraPreview;

import android.os.Bundle;

import com.getcapacitor.BridgeActivity;
import com.getcapacitor.Plugin;
import java.util.ArrayList;

public class MainActivity extends BridgeActivity {
  @Override
  public void onCreate(Bundle savedInstanceState) {
    super.onCreate(savedInstanceState);

    // Initializes the Bridge
    this.init(savedInstanceState, new ArrayList<Class<? extends Plugin>>() {{
      // Additional plugins you've installed go here
      // Ex: add(TotallyAwesomePlugin.class);
      add(CameraPreview.class);
    }});
  }
}
```

Logic file: 
```
  openCamera() { 
    const cameraPreviewOptions: CameraPreviewOptions = { 
      position: 'rear',
      parent: 'cameraPreview',
      className: 'cameraPreview',
      disableAudio: true,
      width: 1920, 
      height: 1080

    }; 

    CameraPreview.start(cameraPreviewOptions); 
    this.cameraActive = true; 
  }
```

**Smartphone (please complete the following information):**
[image]

Oleg-Kasianets-3DLOOK, 2021-09-30:
I have the same problem!

rymeier, 2021-10-01:
I have a similar issue but my Camera Preview eventually launches but that video icon flashes for a few seconds first.

rymeier, 2021-10-01:
@ELHoussineT I had to add the toBack: true, property to the options to get the video to show up properly on Android behind the HTML overlay.

```
Plugins.CameraPreview.start({
              parent: "content",
              toBack: true,
              position: "front"
            });
```

pbowyer, 2021-12-16:
I'm closing this as it does work, but takes some fiddling to get it to show up. We need to improve the documentation. Pull requests are welcome 👍

closed 2021-12-16 by pbowyer