capacitor-community/camera-preview #150: Compilation error [Interface 'CapacitorInstance' incorrectly extends interface 'CapacitorGlobal'.]

ncham, 2021-06-18:
**Describe the bug**
I get this error when compiling my ionic, angular project.

**To Reproduce**
"@capacitor/cli": "^2.4.7",
"@capacitor/core": "^2.4.7",
"@capacitor-community/camera-preview": "^1.2.1",

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: [windows 10]
 - Browser [Chrome]

**Additional context**
Add any other context about the problem here.

ncham, 2021-06-18:
After adding this issue only I saw others also had the same problem.
Sorry for adding a duplicated issue, hope you guys will fix this soon. Thanks in advance.

Julias0, 2021-06-21:
For a quick fix add
```
 "skipLibCheck": true
```

to tsconfig.json under compilerOptions

Kaoschuks, 2021-07-07:
I have created a repo here that uses this plugin and @Julias0  answer and other basic capacitor methods for getting camera data

https://github.com/Kaoschuks/ionic-media-editor

Please review

pbowyer, 2022-03-06:
I'm closing as we now target Capacitor 3. If this issue happens again please open a new ticket.

closed 2022-03-06 by pbowyer