capacitor-community/camera-preview #151: Capacitor 3 Not Compatible

meforce, 2021-06-24:
Capacitor 3 Not Compatible. Will this support?

darbanhandrew, 2021-07-14:
Hi, I found a work around to use this in the web, but i am a newbie in git i don't know how to share it with you

meforce, 2021-07-14:
Hi, mertcetin.official@gmail.com you can mail.

darbanhandrew, 2021-07-27:
i am so sorry for late answering, do you still need it ? i can email it now.

meforce, 2021-07-28:
> i am so sorry for late answering, do you still need it ? i can email it now.

Yup. please send. I am waiting

meforce, 2021-08-05:
### Solved
I realized there is a simple solution.

[image][image]

**package.json > Capacitor 3+**
```
"@capacitor-community/camera-preview": "^1.2.1",
"@capacitor/android": "^3.0.1",
"@capacitor/cli": "^3.0.1",
"@capacitor/core": "^3.0.1",
```
**Use latest version**
`npm install @capacitor-community/camera-preview@latest`

Capacitor Camera Preview Changes in Connections

```
import { Plugins } from "@capacitor/core"
const { CameraPreview } = Plugins;
import { CameraPreviewOptions, CameraPreviewPictureOptions, CameraPreviewFlashMode } from '@capacitor-community/camera-preview';
```

Other codes can remain the same.

I added **androidmanifest.xml** accesses and it works fine.

```
<!-- Permissions -->
<uses-permission android:name="android.permission.CAMERA" />
<uses-feature android:name="android.hardware.camera" />
<uses-feature android:name="android.hardware.camera.autofocus" android:required="false" />
```

[image]

meforce, 2021-08-24:
This workaround is unofficial workaround

closed 2021-08-24 by meforce