capacitor-community/camera-preview #152: Camera preview on iOS black screen / Android works fine

gamanuel, 2021-06-26:
When I start the camera on iOS it shows a black screen but if I take the picture it shows the picture successfully.
On Android all work as expected

[image]
[image]

Iphone 11 pro max 
Last iOS

gamanuel, 2021-08-08:
I found that the problem was when i had dark mode enable on ios, my solution was to remove or comment all the styles for the dark mode from ionic variables.scss

pbowyer, 2021-12-16:
@gamanuel is correct, Ionic's dark mode needs some workarounds.

hanani14, 2022-01-04:
> I found that the problem was when i had dark mode enable on ios, my solution was to remove or comment all the styles for the dark mode from ionic variables.scss

I already comment all the styles for the dark mode in variables.scss, but still face this problem.  @pbowyer @gamanuel ,there have any other solution on this?