capacitor-community/camera-preview #153: IOS build abort during Swift Compile

oneclickwebstudio, 2021-06-27:
**Describe the bug**
Ios build fails on CapacitorComunityCamera 

**To Reproduce**
Steps to reproduce the behavior:
Click on Product -> Build

**Screenshots**
[image]
[image]

**Errors**
```
Value of optional type 'WKWebView?' must be unwrapped to refer to member 'isOpaque' of wrapped base type 'WKWebView'
Value of optional type 'WKWebView?' must be unwrapped to refer to member 'backgroundColor' of wrapped base type 'WKWebView'
Value of optional type 'WKWebView?' must be unwrapped to refer to member 'scrollView' of wrapped base type 'WKWebView'
Value of optional type 'WKWebView?' must be unwrapped to refer to member 'superview' of wrapped base type 'WKWebView'
Value of optional type 'WKWebView?' must be unwrapped to a value of type 'WKWebView'
Value of type 'CameraPreview' has no member 'videoUrl'
Cannot find 'configureVideoOutput' in scope
Value of type 'CameraController' has no member 'videoOutput'
'nil' requires a contextual type
....and  more...
```
**Additional context**

```
OS: MacOS 11.4
Xcode Version 12.5 (12E262)
Swift 5
@capacitor/ios 3.0.2
@capacitor/cli 3.0.2
@capacitor/core 3.0.2
@capacitor/android 3.0.2 
@capacitor-community/camera-preview: "^1.2.1",

```

Add any other context about the problem here.

oneclickwebstudio, 2021-06-27:
Also if i fix by Xcode recommendation and comment everything where VideoCapture called. App brakes. showing only white screen

nielsvanharen, 2021-07-13:
i have the same issue when trying to compile in xcode (capacitor 3)

Scaalees, 2021-08-18:
Any news on this one? I've just ran into the same problem.

```
"@capacitor-community/camera-preview": "^1.2.1",
"@capacitor/cli": "3.1.2",
"@capacitor/ios": "3.1.2",
```

oneclickwebstudio, 2021-08-18:
> Any news on this one? I've just ran into the same problem.
> 
> ```
> "@capacitor-community/camera-preview": "^1.2.1",
> "@capacitor/cli": "3.1.2",
> "@capacitor/ios": "3.1.2",
> ```

in Xcode `CameraController.swift` file comment lines:

```
136    //                try configureVideoOutput()

406   //        videoOutput!.startRecording(to: fileUrl, recordingDelegate: self)
407   //        self.videoRecordCompletionBlock = completion

415   //        self.videoOutput?.stopRecording()

583   //            self.videoRecordCompletionBlock?(outputFileURL, nil)

585   //            self.videoRecordCompletionBlock?(nil, error)

```

Cuz videoRecording is not yet working on this plugin

Scaalees, 2021-08-20:
Thanks for that! Although I couldn't find anything for that.

I ended up switching camera-preview to `2.0.0-beta.0`. Once done, I ran:

```
ionic cap sync ios
cd ios/App
pod install
```

In Xcode I got this error:

[image]

When clicking, you get something like this:

[image]

Then clicking preform change I could finally deploy the app to my iPad and archive and distribute.
_However you do need to do this everytime you want to test through xcode_

Hope this helps anyone else that runs into this.